riscvarchive/riscv-v-spec #699: further opcode allocation packing

jnk0le, 2021-06-24:
Looking at inst-table.adoc I see that some instructions already share funct6 but there are many more instructions that don't, even though they could be optimized in the same way. 

Currently there are
- 11 `VX` only in OPI
- 4   `XI` only in OPI
- 19 `V` only in OPM
- 3   `X` only in OPM
- some empty space in both

Looking at the instructions the difference between OPI and OPM seems to be the lack of `I` form in OPM.
I think that the some of those can also be packed together like already done with `vrgatherei16`/`vslidedown` or `vsmul`/``vmv`.

kasanovic, 2021-06-25:
The current encoding was chosen to reduce decode complexity, and the instructions packed together had similar functionality.  No concrete improvement has been suggested here.

David-Horner, 2021-06-25:
@jnk0le does raise the concern that opcode space is sparsely allocated.

There is concern among some, myself included, that 32 bit instruction formats are limited and their use requires substantial justification. Some may cry "chicken Little", but I share with @jnkole the concern that no encodings be wasted.

My suggestion is to provide more commentary in the text that justifies the V-extensions extensive use of opcode space. Krste and the TG have done a great jiob in providing ageneralized mechanism that provides a substantially different facility over the base "I".

It provides a feature that can replace scalar code with vectorized logic. even transparently to the source code.
It provides an extensive register cluster more flexible and expansive than the scalar float extensions [F,G,and Q]
It is inherently polymorphic, managing F,G.Q and H datatypes without explicit reference.
Full polymorphism is not precluded by the design. It was intended from the beginning, with this release providing a base that can be readily implemented as performant. Future extensions can "borrow" the I and F encodings to apply to tagged polymorphic datatypes.
It does this with less opcode use than the float extension uses.
It is extensible without increased opcode use. Agnostic tail and mask were incorporated without any increased opcode space.
The investment in V will yield returns in future endeavours including crypto, auto-vectorization, and more data-types, including hypercomplex.

jnk0le, 2021-06-25:
If it's optimized for decoding, then it's not that much of a problem.

But following patterns like `slideup`/`slide1up`/`fslide1up`, shouldn't the vrgatherei16 be in `001100` of OPM (match the vrgather)?
Or `vmsbc` to `010001` of opm and `vsbc` below.
Or `VMUNARY0` one point higher.
Or `vdiv`/`vdivu`/`vfdiv`/`vfrdiv` moved somewhere else and put here `vssubu` and `vssub`.
etc.

`vmv<nf>r` doesn't have anything to pattern with, especially vsmul. (definitely looks like opspace optimization)
Also some things like `vmsgt` don't pattern with equivalent `vmfgt`.

kasanovic, 2021-09-18:
vrgatherei16 is only one bit away from vrgather (i.e., equidistant in encoding to your suggestion). The placement reuses an otherwise difficult to use slot in encoding table (while your suggestion would tie up a V/X pair).

Float and integer compares have only very limited hardware connection, so shifting in encoding would be only aesthetic, and not 1-1 either.

In general optimization of encoding choices involves a trying to match common features in multiple binary dimensions for multiple instruction types prioritizing features that impact circuit latencies (e.g.,  use of scalar register), constrained by available space, and with historical allocations that if moved would cause ripple effects. 

We won't be changing these encodings at this point, as there is no discernible advantage, so closing the issue.

closed 2021-09-18 by kasanovic