riscvarchive/riscv-v-spec #713: add weakly ordered indexed store

jnk0le, 2021-07-14:
extracted a single point from #698 

As in the case of unordered stores, the accesses can be observed out of order, but it will have the property of storing only the "oldest" at given index (as observed by subsequent execution). Final result will match the one of regular indexed ordered store.

The indexed-ordered stores can then remain according to normative notes, without the x86 like trickery.

In case of cheap implementation it can reuse logic from regular indexed ordered stores.

kasanovic, 2021-07-15:
I can't see this potential optimization justifying adding a whole another set of indexed store instructions.

I see that in many cases, the ordered indexed store will only be consumed by the local hart, or will have shared data properly synchronized between harts, so remote harts shouldn't care about element order within a single instruction, but the requirement that the local hart sees them in element order - and that remote harts must see them in that same order once instruction has completed - probably prevents useful optimizations for remote harts in practice.

jnk0le, 2021-07-15:
Now I see that there is nowhere to put this new mode, except probably `mew` in VSX* encoding. 128+ indexes are not very usefull, but that might disturb decoders a bit.