riscvarchive/riscv-v-spec #716: More specific description about the setting of vill bit

jovireal, 2021-07-29:
the curent description of vill setting is very short in 3.3.4：
”The vill bit is used to encode that a previous vset{i}vl{i} instruction attempted to write an unsupported value to
vtype.”
How to judge if the value is unsupported？based only on sew/vlmul/vma/vta，or the reserved bit should also be considered？
 
Besides，for vsetvl，at least two cases will set vill，one is that the sew/vlmul/vma/vta combination in rs2 is not supported ( the same as vsetvli)，one is that the vill bit in rs2 is set.  Or maybe a non-zero reserved bit in rs2 will also set vill？
But the spec only writes "When the vill bit is set, the other XLEN-1 bits in vtype shall be zero." which is not easy to get the point.

kasanovic, 2021-08-02:
Portable software can only make use of features in the relevant platform profile, but assuming it does so, the software will not have to check vill.  If there is a mistake in the software, it will cause a trap when attempting to execute a vector instruction with an illegal configuration.

The current (terse) quoted spec does capture the other cases mentioned.
It subsumes the case where a non-zero reserved bit is set and the implementation does not support that bit.
It also subsumes the case where vill is set in vtype argument, as inherently if the vill bit is set, the configuration is not supported.

jovireal, 2021-08-03:
Yet these two cases are not explicitly explained in the spec. 
Implementations might choose to ignore the non-zero reversed bit(like other CSRs) and only base on SEW and LMUL to set vill，as this is the only case made clear in the spec “An attempt to set an unsupported SEW and LMUL configuration sets the vill bit in vtype.”

kasanovic, 2021-09-14:
The spec does say any unsupported value will set vill, and is referring to the whole vtype value.  The sentence about SEW/LMUL is in section discussing those specific fields only. I will add clarifications.

commit 2046b924 (2021-09-14): Clarified handling of illegal vtype values.
Closes #716.
closed 2021-09-14 by kasanovic