riscvarchive/riscv-v-spec #717: how can we get the last scaler data element ?

figozhang, 2021-07-30:
i am confusing how to get the last scaler vector element in a vector register? for example, if we want to use V instructions to implement the strcmp() function.

int strcmp(const char* str1, const char* str2) {
	char c1, c2;
	do {
		c1 = *str1++;
		c2 = *str2++;
	} while (c1 != '\0' && c1 == c2);
	
	return c1 - c2;
}

so the c1 and c2 is the last scaler vector element, we can use vmsbf instruction to setup the mask, but i don't know how to get the data content to a general register?

aswaterman, 2021-07-30:
There are many valid strategies for vectorizing strcmp that have been proved out using RVV. I suggest looking at existing examples for strlen etc. and then spending some time thinking about it.

figozhang, 2021-07-30:
hi Aswaterman，

i guess your suggestion is we can use the Vector Mask Instructions like vmfirst  to know which data element is met the condition like (c1 != '\0' && c1 == c2) in strcmp() function, and then read the last data from memory.

but my concern is that the last data has readed into vector register, why not we provide a register for developer to get it from vector register?  Re-read the data from memory is very slow than fetch from vector register.

i saw ARM SVE also provide some instructions to read the data of vector register to general register, for example, LASTA instruction.

and an other thing is the, if we want to calculate 4x4 matrices, like A x B =C
load the A data into V0， load B data into V1，V2 for C.

for example, if we want to calculate this:

c0 + = a0*b0，c1 +=a1*b0, c2 +=a2*b0,   c3 +=a3*b0, 

we can use this instruction in RVV:

flw ft0, (t1)
vfmacc.vf v2, ft0, v0

but the ft0，should be re-load from memory, right?
my point is that, the b0 data has loaded into V1， why not provide a instruction to read it from v resigter?

In ARM SVE, this instruction  like this:
fmla v2.4s, v0.4s, v1.s[0]

it can directly use the scaler data to FMLA instruction operation.

aswaterman, 2021-07-30:
The redundant load is an extremely minor cost.

Just because another ISA offers an instruction doesn’t make it a good idea…

figozhang, 2021-07-30:
you mean the data will be in cache, if it is cache miss or this data of cache line evict from cache, it will be re-load from memory.

aswaterman, 2021-07-30:
Yes, I mean that the reloaded data is nearly certain to be cache hit, and so the cost to re-access it is very low.

Overall, the fact that RVV scalars are sourced from scalar registers instead of vector registers is a feature, not a bug: doing so substantially reduces register pressure by freeing up vector registers to hold vectors.  Furthermore, the most common source of scalar operands to vector instructions is from other scalar instructions, and so directly using the scalar registers reduces operand traffic.

In the less-common case that a vector operand need be broadcast, we recommend using `vrgather.vi` or `vrgather.vx`.

figozhang, 2021-07-31:
Thanks very deeply explanation, sounds interesting!

kasanovic, 2021-08-02:
I just quickly sketched out the strcmp routine below (not tested, so probably has bugs).

````
    .text
    .balign 4
    .global strcmp
  # int strcmp(const char *src1, const char* src2)
strcmp:
    ##  Using LMUL=2, but will work for larger LMULs
    li t1, 0                # Initial pointer bump
loop:
    vsetvli t0, x0, e8, m2, ta, ma  # Max length vectors of bytes
    add a0, a0, t1          # Bump src1 pointer
    vle8ff.v v8, (a0)       # Get src1 bytes
    add a1, a1, t1          # Bump src2 pointer
    vle8ff.v v16, (a1)      # Get src2 bytes

    vmseq.vi v0, v8, 0      # Flag zero bytes in src1
    vmsne.vv v1, v8, v16    # Flag if src1 != src2
    vmor.mm v0, v0, v1      # Combine exit conditions
    
    vfirst.m a2, v0         # ==0 or != ?
    csrr t1, vl             # Get number of bytes fetched
    
    bltz a2, loop           # Loop if all same and no zero byte

    add a0, a0, a2          # Get src1 element address
    lbu a3, (a0)            # Get src1 byte from memory

    add a1, a1, a2          # Get src2 element address
    lbu a4, (a1)            # Get src2 byte from memory

    sub a0, a3, a4          # Return value.

    ret
````
A hypothetical "move element from vector register index to scalar" could admittedly be a bit faster on some possible microarchitectures than the scalar loads to memory, but could also be a lot slower on others, and would add the complexity of a new data path and new data-dependent hazard to the pipeline.

If someone wants to test this, we could add to examples directory.

figozhang, 2021-08-02:
i have test your strcmp_asm code, it works fine.

i wrote a simple test case on it.

int strcmp(const char* str1, const char* str2) {
	char c1, c2;
	do {
		c1 = *str1++;
		c2 = *str2++;
	} while (c1 != '\0' && c1 == c2);
	
	return c1 - c2;
}

char *str1 = "i am learning riscv now, riscv is very good for me!!";
char *str2 = "i am learning riscv now, riscv is very good for us!!";

int main(void)
{
	int ret; 

	ret = strcmp(str1, str2);
	printf("C: ret = %d\n", ret);

	ret = strcmp_asm(str1, str2);
	printf("ASM: ret =%d\n", ret);
}

Here is the run result.
spike --varch=vlen:128,elen:32,slen:128 --isa=rv64gcv pk test
bbl loader
C: ret = -8
ASM: ret =-8

figozhang, 2021-08-02:
in this instruction，
vsetvli t0, x0, e8, m2, ta, ma  # Max length vectors of bytes

x0 should be initial as -1 ？

figozhang, 2021-08-02:
An other one is we has better to using temporary register for a0 and a1 to access the memory, otherwise other invalid introduction will change the content of source and destination string.
for example:
	mv a2, a0             # str1
	mv a3, a1             # str2

kasanovic, 2021-08-02:
> in this instruction，
> vsetvli t0, x0, e8, m2, ta, ma # Max length vectors of bytes
> 
> x0 should be initial as -1 ？

x0 is fixed zero register and when used in vsetvli in this way sets the maximum vector length - see spec for vsetvli

> An other one is we has better to using temporary register for a0 and a1 to access the memory, otherwise other invalid introduction will change the content of source and destination string.
> for example:
> mv a2, a0 # str1
> mv a3, a1 # str2

not needed given in standard calling convention argument registers are caller save

kasanovic, 2021-08-02:
I'm going to add to examples directory given that it at least compiled and worked for one test case.
Thanks for trying it out.

commit d19a1053 (2021-08-02): Added strcmp example.
Closes #717.
closed 2021-08-02 by kasanovic