riscvarchive/riscv-v-spec #719: shuffles and other horizontal operations

JishinMaster, 2021-08-05:
Dear all,
I work on an open source library called simd_utils (https://github.com/JishinMaster/simd_utils), which gives some optimised mathematical functions using SIMD/Vector intrinsics.

I have a few functions using riscv vector extensions and planning to do more.
I have trouble finding how to replace sse/neon instructions which do shuffle or horizontal add with riscv vector ones.

Is there a way to do so? Would dedicated instructions be included in a future revision?

jnk0le, 2021-08-05:
Shuffling can be done by `vrgather` or `vrgatherei16` (for VLA corectness).

There are horizontal reductions (e.g. sum all elements).
"even-odd horizontal" operations are replaced by widening.

jnk0le, 2021-08-05:
Dot products etc. are not available yet as those were postponed with ediv.