riscvarchive/riscv-v-spec #720: Assembler names for LMUL collide with potential matrix register names

LDelhez, 2021-08-16:
The current spec suggests the following assembler names for the LMUL setting in `vset{i}vli` instructions:
```
mf8    # LMUL=1/8
mf4    # LMUL=1/4
mf2    # LMUL=1/2
m1     # LMUL=1
m2     # LMUL=2
m4     # LMUL=4
m8     # LMUL=8
```

However, `m1`, `m2`, `m4` and `m8` are ideal names for matrix registers in a potential future RISC-V matrix extension. 

Would it not be wiser to change the LMUL assembler names in "V" to prevent any confusion in future codes that will require both vector and matrix registers?

Using a word with multiple letters would solve this issue. For instance, `m2` could become `mi2` (where `i` stands for _integer_). 

---
P.S.: To some degree, a similar name collision issue could also exist with the SEW settings (e.g. `e1`).

LDelhez, 2021-08-16:
Of course, the _assembler program_ will always know whether a symbol is a matrix register or an `LMUL` setting. I am only pointing out the fact that having one symbol that can correspond to two different things makes the assembly code less readable _for a human_.

Here, this confusion could be easily avoided.

aswaterman, 2021-08-16:
Definitely not _easily_ avoided: changing the `vsetvli` syntax would be extremely disruptive at this point.

In any case, I don't think the cognitive load on the assembly programmer is significant.  Humans also need to read these things in context, and, to borrow Kito's example, they already have to understand that `la t0, m1` is not a reference to a register named `m1`.

nick-knight, 2021-08-16:
> la t0, m1

Wait, `t0`, is that tensor register zero? /s