riscvarchive/riscv-v-spec #721: Clarification about the vset{i}vl{i} instructions (Section 6.1)

just-for-fun-too, 2021-08-16:
In Section 6.1, it says"
"**Use of the instruction with a new SEW/LMUL ratio that would result in a change of VLMAX is reserved...**" 
The ratio will get changed over time, as different vector sets are used for different computations. 
The question is what else is required in order to make the change? In other words between changing this ratios what needs to be done (set?cleared?...) and where is this specified?

aswaterman, 2021-08-16:
That statement applies only to the variant that uses rs1=x0 and rd=x0.  It's legal to change the SEW/LMUL ratio when rd != x0 or rs1 != x0.

closed 2021-08-16 by aswaterman

just-for-fun-too, 2021-08-17:
Thank you @aswaterman!

I really feel Section 6.1 is somewhat confusing. At least to clarify this specific issue if it could be said (addition in bold):

"Use of the  instruction **when rs1=x0 and rd=x0** with a new SEW/LMUL ratio that would result in a change of VLMAX is reserved."

would be helpful.