riscvarchive/riscv-v-spec #725: FPR reuse the lower part of VPR

jovireal, 2021-08-31:
when both vector extension and F extension are supported, is it possible to reuse the lower part of vector register for floating-point register? As in hardware design, scalar FP operation and vector FP operation usually share the same issue pipe and computation logic for area efficiency. by reusing the register source, the operand mux, the read port/write port logic and the result fanout can be largely reduced.

jnk0le, 2021-08-31:
0.6 vectors were shared with fp. Now it's separate. 

Sharing pipeline for scalar and vector can also have disadvantages. Like not being able to perform scalar and vector operation in paraller or not having 'extra' read port for scalar-vector ops.

kasanovic, 2021-09-13:
There was extensive discussion over course of development leading to current design.  Spec section 5.1 Scalar operands has some rationale on this.

closed 2021-09-13 by kasanovic