riscvarchive/riscv-v-spec #729: zvl* and zve* extension names are invalid per the current ISA manual

asb, 2021-09-03:
The current ISA manual [states](https://github.com/riscv/riscv-isa-manual/blob/5bce9789a9183f553ca444d820e9152bde442c54/src/naming.tex#L79) "Standard extensions can also be named using a single `Z` followed by an alphabetical name and an optional version number.".

The zvl* and zve* extensions don't meet this criteria due to the embedded numbers. I'm not sure there's a particularly elegant solution to this, as the convenience of embedding numbers in these extension names is obvious. To allow these names, the ISA manual would need to be updated to allow extension names with embedded numbers provided they ended with a letter of the alphabet other than p.

jrtc27, 2021-09-03:
Ending with a p is fine, it's the specific case of ending with `[0-9]p` and following it with just a major version number that's ambiguous (omitting the version entirely is unambiguous, as is having both major and minor version numbers, it's just the middle case that's ambiguous, but even if that weren't the case it'd probably be a terrible idea to allow extensions to end with `[0-9]p`). "...0pa" and "...0ap" are both unambiguous, though potentially still a little confusing (but then again not really more so than Zvl32b which at first glance someone might take to mean version 32.0 of Zvl, plus the ratified version of B).

kito-cheng, 2021-09-03:
I created a similar issue before, and created a pr for isa spec try to update rule.
https://github.com/riscv/riscv-v-spec/issues/697
https://github.com/riscv/riscv-isa-manual/pull/718

kasanovic, 2021-09-18:
To be revisited after spec goes out, but yes plan is to extend naming in backwards-compatible way by requiring non-p letter after numeric digits.