riscvarchive/riscv-v-spec #730: `misa.v` for `zve*` extensions?

kito-cheng, 2021-09-06:
Dose `misa.v` will be set if `zve*` extension is available?

Some software like linux kernel and glibc use that to determine the vector extension is enabled or not, so it would be problem if `misa.v` is not set for `zve*`, but it's kind of violate the definition in misa.

```
The Extensions field encodes the presence of the standard extensions,
with a single bit per letter of the alphabet (bit 0 encodes presence
of extension ``A'' , bit 1 encodes presence of extension ``B'',
through to bit 25 which encodes ``Z''). 
```

Potential options:
- Do not set `misa.v` for `zve*` and wait https://github.com/riscv/configuration-structure to resolve the detection issue?
- Set `misa.v` for `zve*`, and add new rule for misa.v in priv spec.

lbmeng, 2021-09-09:
How could Linux kernel or glibc determine V-extension via misa.v which is a M-mode CSR? And I don't see SBI calls defined for getting misa too.

aswaterman, 2021-09-10:
I think the normal expectation is that the relevant information is passed in the DeviceTree.

kito-cheng, 2021-09-10:
I guess I am confused by qemu implementation https://github.com/qemu/qemu/blob/master/linux-user/elfload.c#L1446 .

Regardless linux and glibc, this issue still wroth to clarify I think?

jim-wilson, 2021-09-10:
The qemu implementation is just reading a single bit, and passing it to glibc, but that is unlikely to be good enough.  Your C API pull request says we need 3 values: min vector length, max int element width, and max FP element width.
[https://github.com/riscv-non-isa/riscv-c-api-doc/pull/21](url)

it isn't possible to get these 3 values from the single misa.v bit.  And passing 3 integers via the aux vector isn't as easy as passing down a single bit in AT_HWCAP.

Hence the suggestion to put vector info in the DTS for the linux kernel.  For glibc, we might want RISC-V specific additions to the aux vector, or else try to encode the info and stuff it into AT_HWCAP2 which is currently unused for RISC-V.

lbmeng, 2021-09-10:
> Regardless linux and glibc, this issue still wroth to clarify I think?

Yes, this issue is still valid :) It's just the reason that confused me.

kito-cheng, 2021-09-11:
> Hence the suggestion to put vector info in the DTS for the linux kernel. For glibc, we might want RISC-V specific additions to the aux vector, or else try to encode the info and stuff it into AT_HWCAP2 which is currently unused for RISC-V.

I think it would be an important topic to discuss on linux plumber :)

kasanovic, 2021-09-18:
For consistency with previous standards, the misa.V bit will be set for the single-letter V extension, but not for others.

https://github.com/riscv/riscv-v-spec/commit/1bfee80bbb5415bf25641bcc014e0ffc24d2e451

closed 2021-09-18 by kasanovic