riscvarchive/riscv-v-spec #739: Non-rounding averaging add and subtract

dsharletg, 2021-09-22:
I noticed that in the v1.0 spec, rounding add and subtract averaging instructions are present (`vaadd[u]` and `vasub[u]`), but there do not appear to be any non-rounding versions. Both are very useful, because they can be used in trees to produce averaging trees of minimal bias. For example, `(a + 2b + c + 2)/4` = `(b + (a + c)/2 + 1)/2` = `vaadd(b, missing_non_rounding_averaging(a, c))`. With only the rounding `vaadd`, we can only compute `(a + 2b + c + 3)/4` = `vaadd(b, vaadd(a, c))`, which has worse error.

nick-knight, 2021-09-22:
Is it possible to vary the fixed-point rounding mode
https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#38-vector-fixed-point-rounding-mode-register-vxrm
to simulate the desired behavior?

dsharletg, 2021-09-22:
I didn't realize the roundoff_signed functions were configurable. However, I still don't think that helps. We have large trees of these averaging ops to implement particular kernels, with the rounding mode of each one varying and carefully chosen to minimize bias and error. It seems like it would be slow/impractical to change the rounding mode frequently (possibly for each individual instruction).

nick-knight, 2021-09-22:
Let me try asking a different way. What are the precise semantics of your proposed `missing_non_rounding_averaging(a, c)`?

dsharletg, 2021-09-22:
We would basically just want `ceil_avg(a, b) = (a + b + 1)/2` and `floor_avg(a, b) = (a + b)/2`, so we can use both of these together in larger expressions. For example:

```
floor_avg(floor_avg(a, c), b) = (a + 2b + c + 0)/4
floor_avg(ceil_avg(a, c), b) = (a + 2b + c + 1)/4
ceil_avg(floor_avg(a, c), b) = (a + 2b + c + 2)/4
ceil_avg(ceil_avg(a, c), b) = (a + 2b + c + 3)/4
```

nick-knight, 2021-09-23:
Sorry, I'm still not completely clear. You seem to expect rational expressions like `(a+b)/2` to be interpreted as integers, so you must be assuming some kind of rounding rule when `a+b` is odd. Here are four possibilities in that case:

- Round to positive infinity. This can be implemented by `vaadd` with `vxrm == rnu` 
- Round to negative infinity. This can be implemented by `vaadd` with `vxrm == rdn`
- Round towards zero. I don't know a single-instruction sequence. 
- Round away from zero. I don't know a single-instruction sequence. 

I admit I don't have experience in computing "averaging trees of minimal bias". I'll step back and let others attempt to tackle your question.

dsharletg, 2021-09-23:
In my examples, `floor_avg` is equivalent to `vaadd` with `vxrm == rnu`, and `ceil_avg` is equivalent to `vaadd` with `vxrm == rdn`.

The issue is I think we would need to flipping the rounding mode very frequently, which seems like a performance issue. In general, statefulness like this in instructions is quite a bit of a pain to deal with IMHO.

dsharletg, 2021-09-23:
I edited the previous comment to just use shifts (which I assume to have no clever rounding, just truncating).

dsharletg, 2021-09-23:
I see now that shifts are also affected by this rounding register... :( IMO, this really makes the instruction set harder to use for a lot of the code I'm used to writing on ARM and other targets.

nick-knight, 2021-09-23:
Toggling of the `vxrm` CSR is performed by scalar instructions, which can often be scheduled far in advance of the associated vector instructions. This is comparable to changing the vector length (`vl` CSR), and the overheads of that have been discussed at length elsewhere. Moreover, although I don't know the details of your use-case, I suspect the pattern of `vxrm` changes are often known statically, so can be encoded by an immediate (`csrwi`). I believe this property could be particularly to relevant out-of-order implementations.

I certainly agree that it would reduce static and dynamic code size if `vxrm` were encoded directly in the instruction. However, in my opinion, it is unsound reasoning to conclude a performance issue based solely on instruction count, at least without a particular implementation in mind.

dsharletg, 2021-09-23:
It's not really just about performance. There's also the practical matter of it just being kind of a pain to have global state and instructions that are not pure functions:
- Assembly is harder to read. To know the behavior, you have to find the last time the control register was set, which may be before some control flow.
- This is an extra constraint for the compiler and HW to consider when attempting to reorder instructions.
- The only reasonable thing for compilers to do is emit code to change the state register every time one of the affected instructions is used (and then maybe try to optimize redundant setting of the state registers?).

nick-knight, 2021-09-23:
I completely agree with your practical concerns. This was already an issue for scalar floating-point's dynamic rounding mode and exception flags (`frm` and `fflags` CSRs).

Almost every vector instruction depends on global state --- the `vl` and `vtype` CSRs --- and this is already a major headache for the [C intrinsics](https://github.com/riscv-non-isa/rvv-intrinsic-doc). The current LLVM implementation emits a state change (`vset{i}vli`) before every instruction, and then performs a succession of dizzying passes in an attempt to elide redundant ones. I spend a fraction of my working hours filing tickets about corner-cases where LLVM's `vset{i}vli`-elision fails. So your point certainly resonates with me.

As far as I'm aware, the C intrinsics interface hasn't even agreed on how to handle `vxrm`:
https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/36
https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/46
https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/84
https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/27#issuecomment-870713856

kasanovic, 2021-09-23:
Understand and agree with the complaint generally (except that saturation mode is part of instruction encoding).  Rounding mode was put in CSR just for encoding space reasons.  Also note that in a vector machine, the mode change is only single-cycle occupancy whereas vector operation could run for quite a few cycles, so actual execution time overhead can be low/zero. There were requests for more fixed-point support and plan was to add more later as extensions, and these could include static round versions, but these could chew up a lot of encoding space unless carefully selected.  Compiler teams have had success removing extraneous vsetvli instructions using the suggested approach (as Nick mentions), so would definitely try that path.

nick-knight, 2023-05-09:
Agree there is confusion about the term truncation, whose meaning changes when passing from sign-magnitude to two's complement.

Disagree about the relation between RNU and RTZ: they are distinct.