riscvarchive/riscv-v-spec #740: Lacking segment type for whole register load/store

rjiejie, 2021-09-24:
we are glad to see the vector spec add whole register load/store instructions,
they help compiler/gcc to create more efficient code for spilling register without vsetvl* instructions :)

but there are not any instructions for unit-stride segment type ?
I think the this type also have same requirement like normal memory load/store,
so I suggest that we should add instructions for the type like bellow :

# Format
vlseg<nf>e<eew><lmul>.v vd, (rs1), vm      # Unit-stride segment load template
vsseg<nf>e<eew><lmul>.v vs3, (rs1), vm     # Unit-stride segment store template

# Examples
vlseg8e8m2.v vd, (rs1), vm   # Load eight vector registers with eight byte fields in 2 lmul.
vsseg3e32m2.v vs3, (rs1), vm  # Store packed vector of 3*4-byte segments from vs3,vs3+1,vs3+2 to memory in 2 lmul.

aswaterman, 2021-09-24:
These operations can be emulated just by inserting a `vsetvli` (to select the desired LMUL and set `vl` to maximum) and using an existing unit-stride segment load or store.  Plus, it doesn't seem like they would provide much value.

The justification for the whole register loads and stores was to avoid the need to override `vtype`/`vl` in spill/fill code.  But that argument seems irrelevant here, since it doesn't make sense to use segment loads/stores in spill/fill code.

rjiejie, 2021-09-24:
> These operations can be emulated just by inserting a `vsetvli` (to select the desired LMUL and set `vl` to maximum) and using an existing unit-stride segment load or store. Plus, it doesn't seem [...]

I mean gcc use segment loads/stores in spill/fill code actually, so we need to avoid the need to override `vtype`/`vl`.

rjiejie, 2021-09-24:
@kito-cheng any thought for this requirement ?

aswaterman, 2021-09-24:
GCC probably needs to change.  Even ignoring this discussion, some implementations will execute segment loads and stores at lower performance than regular unit-stride loads and stores.  If you need to e.g. spill registers 12 - 16, you should use `vs4r v12` then `vs1r v16`, rather than `vsseg5 v12`.

rjiejie, 2021-09-24:
@topperc How to spill about for unit-stride segment loads and stores in LLVM ?
it's same as @aswaterman mentioned ?

topperc, 2021-09-24:
> @topperc How to spill about for unit-stride segment loads and stores in LLVM ?
> it's same as @aswaterman mentioned ?

LLVM uses multiple whole register load and store instructions. Incrementing a scalar pointer register by VLENB * LMUL between each. We use the LMUL from the segment load/store. We don't optimize to use a larger load/store for multiple registers that are more aligned than the LMUL.

kito-cheng, 2021-09-25:
I think use multiple whole register load and store should be right way to fix that issue on GCC, many of GCC implementation are done before whole register load and store added...and we didn't have put too much engineer there now :(

rjiejie, 2021-09-26:
@aswaterman @topperc @kito-cheng 
Thanks for your feedback,
I will make the patch for GCC also :)

rjiejie, 2021-09-28:
Upload GCC PR: https://github.com/riscv-collab/riscv-gcc/pull/306

closed 2021-11-09 by kasanovic