riscvarchive/riscv-v-spec #749: errors when compiling

PPParticle, 2021-11-08:
I want to run riscv isa v extension in my computer. But I have encountered some errors.

riscv-gnu-toolchain

RISCV=/home/qiao_sh_pudong/RISCV/Bin/riscv64/
RISCV_BIN=$RISCV/bin/
RISCV_ELF=$RISCV/riscv64-unknown-elf/
QEMU_BIN=/home/qiao_sh_pudong/RISCV/Bin/qemu/bin/
LLVM_BIN=/home/qiao_sh_pudong/RISCV/Bin/llvm/bin/
PATH=$PATH:$RISCV:$RISCV_BIN:$RISCV_ELF:$RISCV_ELF/bin:$LLVM_BIN:$QEMU_BIN
$ git clone -b rvv-intrinsic https://github.com/riscv-collab/riscv-gnu-toolchain.git
$ git submodule update --init --recursive 
$ mkdir build
$ cd build
$ ../configure --prefix=$RISCV --with-arch=rv64gcv --with-abi=lp64d
$ sudo make -j6
riscv-pk

$ cd riscv-pk/
$ mkdir build
$ cd build/
$ ../configure --prefix=$RISCV --host=riscv64-unknown-elf --with-arch=rv64gcv
$ make
$ sudo make install
There is no something wrong when installing. But when I git riscv-v-spec and try to run the examples in it.
Some mistakes occur.
$ riscv64-unknown-elf-as --march=rv64gcv -o strcpy.o strcpy.s
It shows that

strcpy.s: Assembler messages:
strcpy.s:9: Error: unrecognized opcode `vsetvli x0,t0,e8,m8,ta,ma'
strcpy.s:10: Error: unrecognized opcode `vle8ff.v v8,(a1)'
strcpy.s:11: Error: unknown CSR `vl'
strcpy.s:12: Error: unrecognized opcode `vmseq.vi v1,v8,0'
strcpy.s:13: Error: unrecognized opcode `vfirst.m a3,v1'
strcpy.s:15: Error: unrecognized opcode `vmsif.m v0,v1'
strcpy.s:16: Error: unrecognized opcode `vse8.v v8,(a2),v0.t'

There are several problems I want to know.

Is there something wrong about my installation process?
Do I use the right way to test the "v" extension? If I am wrong, how can I test?

jim-wilson, 2021-11-08:
The riscv-gnu-toolchain rvv-intrinsic branch is not actively maintained and is badly out of date.  Only LLVM has up to date vector support.  This is in the upstream tree.  Though if you only need an assembler, you can find working GNU as vector support in the upstream FSF Binutils branch called users/riscv/binutils-integration-branch.  There are plans to move that support to master now that the vector extension is frozen for public review.

PPParticle, 2021-11-09:
> The riscv-gnu-toolchain rvv-intrinsic branch is not actively maintained and is badly out of date. Only LLVM has up to date vector support. This is in the upstream tree. Though if you only need an [...]

Thanks for your replying so much.
I understand that rvv-intrinsic branch is not working. 
Does that mean if I want to run "V" extension ISA now, I have to change to llvm complier?
What obout pk and spike simulator?

nick-knight, 2021-11-09:
To the best of my knowledge, the only working implementation of the [C intrinsics for RVV 1.0](https://github.com/riscv-non-isa/rvv-intrinsic-doc) is in LLVM. Please note that the C intrinisics API is a separate project/repository than this one, and so questions about the intrinsics are better directed there. Same with Spike and PK: those are separate projects/repositories as well. I have personally been able to build a C program using RVV intrinsics, compile it with LLVM, and run it on Spike+PK. (I have not been able to build PK with LLVM, but that is a separate issue from RISC-V vectors.)

PPParticle, 2021-11-09:
> To the best of my knowledge, the only working implementation of the [C intrinsics for RVV 1.0](https://github.com/riscv-non-isa/rvv-intrinsic-doc) is in LLVM. Please note that the C intrinisics API [...]

Thanks for your answers. I totally understand what you have told. 
The reason why I ask here is that my final target is to run V extension no matter what I need. 
Since I notice that the the Additional Resources part of README.md talks about spike+pk support v1.0 and that pk needs riscv-gnu-toolchain supports, I think they are all together. If I don't run successfully, there must be something wrong in them.

closed 2022-01-12 by PPParticle