riscvarchive/riscv-v-spec #750: confuse about the allowed sew when lmul=1/4 in 4.2 section

lhtin, 2021-11-10:
There is an example showing **sew=32 and lmul=1/4** from the spec [4.2](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#42-mapping-for-lmul--1) section. But from the spec [4.4](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#44-mapping-across-mixed-width-operations), there is no support for this condition. **So, I want to confirm that the example in 4.2 and other similar(like sew=32 and lmul=1/8) are allowed aways if VLEN*lmul >= 8. But when using mixed-width operations(i.e. same VLMAX different sew), you have to guarantee the min-width element's elmul >= 1/8 and the ratio eew/elmul be same.** Is my understanding correct?

JamesKenneyImperas, 2021-12-22:
The examples in section 4.4 are showing cases where SEW/LMUL combinations _enable 8 elements_ in this machine configuration (VLEN=128b). There are many other legal SEW/LMUL combinations which do not enable 8 elements and are therefore not shown. Selecting SEW=32 and LMUL=1/4 is legal but would enable just one element with this machine configuration.

Legal SEW/LMUL combinations are subject to many constraints, and these are instruction dependent. For example:
1. The minimum EEW for any non-mask vector operand must not be less that the minimum supported SEW (usually 8, but some hardware might force a larger minimum SEW, and for floating-point operands this must be at least 16);
2. The maximum EEW for any non-mask vector operand must not exceed the maximum supported SEW (usually 32 or 64);
3. The minimum EMUL for any operand must be at least 1/8;
4. The maximum EMUL for any operand must not exceed 8;
5. If destination and source EEWs differ, there are constraints about allowed overlaps;
6. A recent change disallows the same vector registers being used as sources with different EEW in one instruction;
7. Some instructions (e.g. vslideup) have additional constraints on overlaps.

lhtin, 2021-12-29:
Thank you for kindly explaining. It clears my confusion.

closed 2021-12-29 by lhtin