riscvarchive/riscv-v-spec #755: a small suggestion for the descriptions of vnclip

kaitoukito, 2021-11-13:
In manual V1.0, Page58
"For vnclip, ... if the result would overflow the destination viewed as a signed integer"
may better be described as
"For vnclip, ... if the result would overflow or underflow the destination viewed as a signed integer"
as it is signed.

aswaterman, 2021-11-13:
That doesn’t match my understanding of common use of the term “underflow”, which normally refers to a number’s magnitude becoming too small, resulting in a loss of precision. By contrast, exceeding the data type’s range is normally called “overflow”, regardless of the direction.

kaitoukito, 2021-11-13:
> That doesn’t match my understanding of common use of the term “underflow”, which normally refers to a number’s magnitude becoming too small, resulting in a loss of precision. By contrast, exceeding the data type’s range is normally called “overflow”, regardless of the direction.

Thanks for your reply:)

closed 2021-11-13 by kaitoukito