riscvarchive/riscv-v-spec #759: EEW=64 support for index load store when XLEN=32

jovireal, 2021-11-18:
In chapter18.2，it clarifies that All Zve* extensions support all vector load and store instructions (Section Vector Loads and Stores), except Zve64* extensions do not support EEW=64 for index values when XLEN=32.

but in chapter7.2，it says ”If the vector offset elements are wider than XLEN, the least-significant XLEN bits are used in the address
calculation.
 
So which of these two rules should be followed when dealing index load store with EEW=64 under a XLEN =32 and ELEN=64 implementation .

nick-knight, 2021-11-18:
The six standard vector extensions (V and Zve*) specify that index-EEW cannot be wider than XLEN, so this Ch. 7.2 rule never applies. It only can apply in the case of a nonstandard vector extension, like one with XLEN=32 and index-EEW loads/stores.

jovireal, 2021-11-18:
@nick-knight thanks for the explanation~

closed 2021-11-18 by jovireal