riscvarchive/riscv-v-spec #762: upper bits for vmv.x.s when SEW>XLEN

3-Monkey, 2021-11-25:
In chapter 17.1,  it says  the upper SEW-XLEN bits are ignored when SEW>XLEN. For example, if SEW=64 and XLEN=32-bit , the result is bit[31:0]. And then if we want to write the 32-bit result to 64-bit register,  should we  keep the  upper 32-bit data or just sign-extended the result?

aswaterman, 2021-11-25:
There seems to be a contradiction in your question: you wrote XLEN=32, but then describe writing the result to a 64-bit register, which is impossible (because the destination register is XLEN bits wide).

jinma-work, 2023-02-15:
Hi @aswaterman ,
I have encountered this problem now, and I hope my program can run on both rv64 and rv32. However, when I encountered the instruction vmv.s.x on rv32, the high 32 bits were ignored, resulting in the return of some unknown data at the high 32 bits instead of the 0 I expected. I think it is possible to zero the high 32 position or expand the symbol?

This is my function:

int64_t vmvxsi64m1_s_nomask_builtin_test (int64_t *x, size_t vl)
{
    vint64m1_t vx;
    vx = vle64_v_i64m1 (x, vl);
    return  vmv_x_s_i64m1_i64 (vx);
}

topperc, 2023-02-15:
Which compiler are you using? clang emulates `vmv_x_s_i64m1_i64` on rv32 with multiple instructions since a single vmv.x.s can only return 32 bits on rv32. https://godbolt.org/z/5zKTbvMba