riscvarchive/riscv-v-spec #768: vl=0 or vstart ≥ vl is reserved for FP in 10.1; defined for all vector-destination instructions in 5.4

kdockser, 2022-01-04:
In section 10.1 it states that vl=0 or vstart ≥ vl is reserved in the case of vector floating-point instructions:
`Use of the frm field when it contains an invalid rounding mode by any vector floating-point instruction, even those that do not depend on the rounding mode, or when vl=0, or when vstart ≥ vl, is reserved.`

However, section 5.4 states that the behavior is defined for all vector instructions
`When vstart ≥ vl, there are no body elements, and no elements are updated in any destination vector register group, including that no tail elements are updated with agnostic values.`

`Note As a consequence, when vl=0, no elements, including agnostic elements, are updated in the destination vector register group regardless of vstart.`

It seems that the sentence in 10.1 should be limited to the case of an invalid rounding mode:
`Use of the frm field when it contains an invalid rounding mode by any vector floating-point instruction, even those that do not depend on the rounding mode, is reserved.`

aswaterman, 2022-01-05:
By my reading, there is no contradiction—the clause from 5.4 just says that no vector registers are written. Those opcodes being reserved in some cases doesn’t run afoul of that. Raising an illegal-instruction exception, for example, doesn’t cause any elements to be updated.

kdockser, 2022-01-06:
Thanks Andrew. While the behaviors can co-exist, I am wondering if this special-casing  of vector floating point when vstart ≥ vl (and consequently when vl=0) was intentional or if this was a vestige from an earlier draft. If it was intentional, does anyone recall the rationale for this decision?

nick-knight, 2022-01-06:
I'm not following this conversation. My reading of the sentence in question that every instruction listed in chapter 13 is reserved when executed under `frm == 101, 110, 111`. The subsequent note in 10.1 seems to provide a rationale: "... to simplify control logic".

aswaterman, 2022-01-06:
> was intentional or if this was a vestige from an earlier draft

Intentional; it simplifies the control logic.

> Perhaps the confusion is which clauses the "or"s are coordinating?

I'll use some em dashes to disambiguate.

kdockser, 2022-01-10:
Thanks for the clarification in the spec, Andrew. The em dashes did the trick:

`Use of the frm field when it contains an invalid rounding mode by any vector floating-point instruction—​even those that do not depend on the rounding mode, or when vl=0, or when vstart ≥ vl--is reserved.`

The confusion had been caused by the association of "or"s.

closed 2022-01-10 by kdockser