riscvarchive/riscv-v-spec #769: question about instructions vmsbf.m, vmsif.m, vmsof.m

liouslee, 2022-01-07:
hi, all
I am not sure what is the behavior of destination inactive masked-off bits for instructions vmsbf.m, vmsif.m, vmsof.m? Does the "xxxx" means we can fill any value to these bits, regardless of vtype.vma? 
I copied the example codes below, which is from section 15.4 of RVV spec 1.0 frozen:
1 1 0 0 0 0 1 1 v0 vcontents
1 0 0 1 0 1 0 0 v3 contents
                       vmsbf.m v2, v3, v0.t
0 1 x x x x 1 1  v2 contents

In addition, whether all mask-result instructions need to fill the mask-off bits according to the vtype.vma policy, such as vector integer compare instructions？
And is it allowed that the implementation choose to only support mask-agnostic and tail-agnostic for mask-result instructions?
Thanks!

nick-knight, 2022-01-07:
Here's my understanding:

> I am not sure what is the behavior of destination inactive masked-off bits for instructions vmsbf.m, vmsif.m, vmsof.m? Does the "xxxx" means we can fill any value to these bits, regardless of vtype.vma?
I copied the example codes below, which is from section 15.4 of RVV spec 1.0 frozen:
1 1 0 0 0 0 1 1 v0 vcontents
1 0 0 1 0 1 0 0 v3 contents
vmsbf.m v2, v3, v0.t
0 1 x x x x 1 1 v2 contents

Section 5.4 says:

> The inactive elements are the elements within the body but where the current mask is disabled at that element position. The inactive elements do not raise exceptions and do not update any destination vector register group unless masked agnostic is specified (`vtype.vma=1`), in which case inactive elements may be overwritten with 1s.

So, if `vtype.vma = 0`, then `xxxx = 0101`, and if `vtype.vma = 1`, then the following four are possible: `xxxx = 0101`, `xxxx = 1101`, `xxxx = 0111`, `xxxx = 1111`.

> In addition, whether all mask-result instructions need to fill the mask-off bits according to the vtype.vma policy, such as vector integer compare instructions？

Yes, mask-result instructions need to obey the `vtype.vma` policy. (There is, however, a wrinkle concerning the `vtype.vta` policy, as discussed in Section 5.4.)

> And is it allowed that the implementation choose to only support mask-agnostic and tail-agnostic for mask-result instructions?
Thanks!

An implementation must support all four combinations of `vtype.{vma,vta}`. The implementation has some freedom in _how_ it implements mask- and tail-agnostic (as discussed in Section 5.4).

liouslee, 2022-01-07:
I think I get it. Thank you.