riscvarchive/riscv-v-spec #778: What's the purpose of using fractional LMUL?

QuarticCat, 2022-02-28:
Hi, I'm learning rvv and feel puzzled about fractional LMUL. I don't see any benefits of using it except that it takes less stack memory when register spills. It doesn't even increase the number of registers. And I feel it will not reduce the context switch overhead either.

nick-knight, 2022-02-28:
Here's an artificial example. Suppose you are already using 31 of the 32 architectural vector registers. Suppose you want to load a single int8 into the 32nd register, then widen it to int16. Since VLEN >= 128, there should be plenty of room. But try doing this without fractional LMUL and without spilling.

QuarticCat, 2022-02-28:
@nick-knight Does this mean that LMUL=1/8 typically will not be used until we call narrowing operations at least three times? It sounds rare.

topperc, 2022-02-28:
> Hi, I'm learning rvv and feel puzzled about fractional LMUL. I don't see any benefits of using it except that it takes less stack memory when register spills. It doesn't even increase the number of registers. And I feel it will not reduce the context switch overhead either.

Fractional LMULs can't realisitically reduce spill size. There's no whole register load/store for fractional LMUL. So a spill may require a vsetvli change in order to spill only a fraction. That increases the compiler complexity of generating the spill.

QuarticCat, 2022-02-28:
@topperc Oh, got it! Then how does the compiler treat spill of LMUL>1?

nick-knight, 2022-02-28:
Here's an example Krste contributed a year ago: https://github.com/riscv/riscv-v-spec/blob/master/fraclmul.adoc

> Does this mean that LMUL=1/8 typically will not be used until we call narrowing operations at least three times? It sounds rare.

Yes, indeed.

Please also note:

> For standard vector extensions with ELEN=64, fractional LMULs of 1/2, 1/4, and 1/8 must be supported.
> [...]
> For a given supported fractional LMUL setting, implementations must support SEW settings between SEW_MIN and LMUL * ELEN, inclusive.

So, for (single-letter) "V" extension, which has SEW_MIN = 8 and ELEN = 64, only the following LMUL and SEW settings are guaranteed to be supported:

* LMUL = 1,2,4,8: SEW = 8,16,32,64
* LMUL = 1/2: SEW = 8,16,32
* LMUL = 1/4: SEW = 8,16
* LMUL = 1/8: SEW = 8

QuarticCat, 2022-02-28:
@nick-knight Thank you!

topperc, 2022-02-28:
> @topperc Oh, got it! Then how does the compiler treat spill of LMUL>1?

There are whole register load/store instructions for LMUL=1, 2,4 , and 8. So the compiler uses those. Fractional LMUL is treated the same as LMUL=1 for spilling.

QuarticCat, 2022-02-28:
@topperc Thank you!

QuarticCat, 2022-02-28:
I have one more question. *The RISC-V Reader* claims that RVV can reduce context switch overhead by enabling only needed registers. But I cannot find `vsetdcfg` here. So this is no longer true, right? Then how does RVV avoid big context now? I know `mstatus.VS` can help. However, it does not seem to reduce the context size, but can only avoid unnecessary switches.

nick-knight, 2022-02-28:
The discussion of the V-extension in the _RISC-V Reader_ is *very* outdated: a new edition is apparently in progress. `vsetdcfg` and `setvl` merged and evolved into the `vset{i}vl{i}` instruction family. The architecture no longer precisely tracks which of the vector registers are actually "needed". `mstatus.VS` is a coarser-grained mechanism that software can use to decide if it is safe to skip saving/restoring the vector context. I think this is equivalent to your observation?

QuarticCat, 2022-02-28:
@nick-knight Thank you very much!

closed 2022-02-28 by QuarticCat

cao-arvin, 2023-07-27:
> Here's an example ~Krste~ Roger contributed a year ago: https://github.com/riscv/riscv-v-spec/blob/master/fraclmul.adoc
> 
> > Does this mean that LMUL=1/8 typically will not be used until we call narrowing operations at least three times? It sounds rare.
> 
> Yes, indeed.
> 
> Please also note:
> 
> > For standard vector extensions with ELEN=64, fractional LMULs of 1/2, 1/4, and 1/8 must be supported.
> > [...]
> > For a given supported fractional LMUL setting, implementations must support SEW settings between SEW_MIN and LMUL * ELEN, inclusive.
> 
> So, for (single-letter) "V" extension, which has SEW_MIN = 8 and ELEN = 64, only the following LMUL and SEW settings are guaranteed to be supported:
> 
> * LMUL = 1,2,4,8: SEW = 8,16,32,64
> * LMUL = 1/2: SEW = 8,16,32
> * LMUL = 1/4: SEW = 8,16
> * LMUL = 1/8: SEW = 8

> Here's an example ~Krste~ Roger contributed a year ago: https://github.com/riscv/riscv-v-spec/blob/master/fraclmul.adoc
> 
> > Does this mean that LMUL=1/8 typically will not be used until we call narrowing operations at least three times? It sounds rare.
> 
> Yes, indeed.
> 
> Please also note:
> 
> > For standard vector extensions with ELEN=64, fractional LMULs of 1/2, 1/4, and 1/8 must be supported.
> > [...]
> > For a given supported fractional LMUL setting, implementations must support SEW settings between SEW_MIN and LMUL * ELEN, inclusive.
> 
> So, for (single-letter) "V" extension, which has SEW_MIN = 8 and ELEN = 64, only the following LMUL and SEW settings are guaranteed to be supported:
> 
> * LMUL = 1,2,4,8: SEW = 8,16,32,64
> * LMUL = 1/2: SEW = 8,16,32
> * LMUL = 1/4: SEW = 8,16
> * LMUL = 1/8: SEW = 8

> 
But the description of "4.1. Mapping for LMUL ≤ 1" in RVV1.0 SPEC, as fllows:
=====================================
Example, VLEN=128b, LMUL=1/4
Byte           F E D C B A 9 8 7 6 5 4 3 2 1 0
SEW=8b    - - - - - - - - - - - - - - - 3 2 1 0
SEW=16b   - - - - - - - - - - - - - - - - - 1 0
SEW=32b   - - - - - - - - - - - - - - - - - - 0
=====================================
So that's wrong? Shouldn't support SEW=32 when LMUL=1/4?

nick-knight, 2023-07-27:
I don't see any inconsistency between the text you quoted from the spec and my earlier comment.

Let's look at it a different way. The largest EEW is 64, so EEW=32 can only be widened once. So EMUL=1/2 should be sufficient. You'd only need EMUL=1/4 if you needed to widen it twice.

cao-arvin, 2023-07-28:
"Implementations must support fractional LMUL settings for LMUL ≥ SEW/ELEN, for the ELEN value at LMUL=1. An attempt to set an unsupported SEW and LMUL configuration sets the vill bit in vtype." 

As above, SEW=32 LMUL=1/4 is illegal vtype，but it's legal in the description of "4.1. Mapping for LMUL ≤ 1" in RVV1.0 SPEC.
I'm confused here.

nick-knight, 2023-07-28:
It's not illegal, it's optional. If the implementation does not support this optionality, it must set `vill`. Software can leverage this to test for the optional feature.

I agree that the example in Section 4.2 is confusing because it demonstrates implementation-defined behavior. I would support a modification of the example or a clarification.

cao-arvin, 2023-08-01:
Got it. Thank you.