riscvarchive/riscv-v-spec #781: Question about the RISC-V RV64X extension

WilliamWangPeng, 2022-03-14:
Hi dear friends,  
    It's an honor to deliver the issue here, I'm doing ray tracing programs, which is a method of graphics rendering that simulates the physical behavior of light. I want to know if RISC-V has already supported graphics extension RV32X or RV64X ?  like ray tracing, or simple ray tracing program.

```
#include <iostream>

int main()
{
    // Image

    const int image_width = 256;
    const int image_height = 256;

    // Render

    std::cout << "P3\n" << image_width << ' ' << image_height << "\n255\n";

    // The pixels are written out in rows with pixels left to right.
    // The rows are written out from top to bottom.
    for (int j = image_height-1; j >= 0; --j) {
        std::cerr << "\rScanlines remaining: " << j << ' ' << std::flush;
        for (int i = 0; i < image_width; ++i) {
            auto r = double(i) / (image_width-1);
            auto g = double(j) / (image_height-1);
            auto b = 0.25;

            int ir = static_cast<int>(255.999 * r);
            int ig = static_cast<int>(255.999 * g);
            int ib = static_cast<int>(255.999 * b);

            std::cout << ir << ' ' << ig << ' ' << ib << '\n';
        }
    }
    std::cerr << "\nDone.\n";
}

```

thank you
best regards
William

SamB, 2022-03-28:
Um ... that's not a raytracer?

kasanovic, 2022-04-15:
Not an issue with vector spec, so closing.

closed 2022-04-15 by kasanovic