riscvarchive/riscv-v-spec #782: Clarification of legal SEW settings for fractional LMUL

JamesKenneyImperas, 2022-03-17:
Section 3.4.2 says this about fractional `LMUL` and `SEW`:

> For a given supported fractional LMUL setting, implementations must support SEW settings between SEW_MIN and LMUL * ELEN, inclusive.

Could you please add a clarification here to state explicitly whether this defines the _entire range_ of supported SEW values or just the _minimum set_? As an example, when `SEW_MIN=8`, `ELEN=32` and `LMUL=1/4`, this sentence implies that only `SEW=8` _must_ be supported; presumably `SEW=16` and `SEW=32` are also optionally legal?

Thanks.

aswaterman, 2022-03-17:
To me, it seems cut and dry that this sentence expresses a lower bound on functionality and doesn’t forbid additional functionality. What alternative wording would you find clearer?

JamesKenneyImperas, 2022-03-18:
How about changing the sentence to say something like:

For a given supported fractional LMUL setting, implementations _must_ support SEW settings between SEW_MIN and LMUL * ELEN, inclusive, and _may optionally_ support other legal settings as described above.

This makes it explicit that additional functionality is permitted.

JamesKenneyImperas, 2022-03-18:
I think that behavior for some other cases of fractional `LMUL` might also benefit from clarification. For example, when `SEW_MIN=8`, `ELEN=64`, `VLEN=256`, what happens if an attempt is made to set `SEW=64` and `LMUL=1/8`? Is that illegal, or legal with `VL=0`, or something else?

Thanks!

aswaterman, 2022-03-30:
> For example, when `SEW_MIN=8`, `ELEN=64`, `VLEN=256`, what happens if an attempt is made to set `SEW=64` and `LMUL=1/8`? Is that illegal, or legal with `VL=0`, or something else?

If it isn't already explicitly stated somewhere, configurations that would result in VLMAX < 1 should be reserved.

JamesKenneyImperas, 2022-03-30:
I assumed this was almost certainly the case, but I don't think it is explicit anywhere that I could see. I think a statement something like this would cover all bases:

For a given supported fractional LMUL setting, implementations _must_ support SEW settings between SEW_MIN and LMUL * ELEN, inclusive, and _may optionally_ support other legal settings as described above. Configurations that would result in VLMAX < 1 are reserved.

Thanks.

aswaterman, 2022-03-30:
That sounds good to me.  Would like approval from @kasanovic, since although the VLMAX < 1 clause seems obvious, it might be perceived to be normative.