riscvarchive/riscv-v-spec #783: Can arbitrary values be written into tail for tail-agnostic?

jlong299, 2022-03-28:
For case of tail-agnostic, can I write arbitrary values into tail elements? Rather than undisturbed or 1s.
My question is from the tail handling of OOO renamed machine: 
If vector register renaming is done at early stage of OOO pipeline, the VL may be not available yet. Then rename stage has to allocate VL_MAX physical registers. As a result, the execution stage needs to take extra cycles to write 1s to tail registers or read old register to fill them. 
If VL can be obtained at stages after renaming and arbitrary values are allowed to be written into tail registers, you can discard the handling of tail elements, save the extra time (write 1s or old values) mentioned above.
Thanks.

aswaterman, 2022-03-28:
The architecture does not permit that strategy.  (There are microarchitectural techniques to write 1s more efficiently than doing so in the obvious way.)

FWIW, there's potentially a security hole in your preferred strategy.  Physical registers can contain information from other processes, and so if you map a physical register then choose not to write it, a subsequent read of the corresponding architectural register could reveal that information.  Of course, there are ways of closing this hole, but the requirements in the ISA spec sidestep the problem.

jlong299, 2022-03-29:
Thanks you so much for the clear explanation!

closed 2022-03-29 by aswaterman