riscvarchive/riscv-v-spec #786: confused about the description of "vl" register in section 3.5 and section 5.4

gxflying, 2022-04-15:
in the section 3.4 it says the "vl" is the **number of the elements**  to be updated with results from a vector instruction which there is nothing with element index 
while in the section 5.4 it says **body(x)     = (vstart <= x < vl)**  which is the element index of the body ?  
so   the "vl"  is a index ?  or  a  count of elemments in the body ?
https://github.com/riscv/riscv-v-spec/blob/master/v-

2henwei, 2022-04-15:
VL literally can mean the index/count, aimed to update 0~VL-1 elements. Non-zero vstart is a special case, only updating (VL - VSTART) elements.

gxflying, 2022-04-15:
thank for your information， so the vl is more a "index" than a "length"  because it is the length only when vastart=0
may the statement in section 3.5 can be preciser ：something like, 
the vl is defined as the max elelemnt index of the valid element, which is also the count of valid elements  when vstart=0

closed 2022-04-15 by gxflying