riscvarchive/riscv-v-spec #795: vd's sign information in integer multiply-add instructions.

Zissi-Lei, 2022-05-08:
In integer multiply-add instructions, for expample, the vwmaccsu, only the sign information of vs1 and vs2 is given, but for vd I can't see it's sign information in spec. Is vd  always treated as a signed number in all integer multiply-add instructions? Thanks for your time.

aswaterman, 2022-05-09:
If either operand is signed, the result is signed. The spec doesn’t explicitly say so because this is implicitly true.

Zissi-Lei, 2022-05-09:
> If either operand is signed, the result is signed. The spec doesn’t explicitly say so because this is implicitly true.

Yes, but I mean that vd is also used as an operand. What's the signed information about vd as an operand?

nick-knight, 2022-05-09:
The addition with vd is performed the same way whether you interpret the bits of vd as encoding a signed (two's complement) or unsigned integer. I think it makes more sense to think of vd (input and output) as a signed integer, but this interpretation doesn't change the semantics of the instruction.

aswaterman, 2022-05-09:
For accumulating operations, the signedness of the accumulator doesn’t matter, because it’s just modular arithmetic. (The signedness of the multiplicands does matter, by contrast.)

aswaterman, 2022-05-09:
But, conceptually, what Nick wrote matches how I think about these operations.