riscvarchive/riscv-v-spec #798: Why dose whole register store has only EEW=8 encoded

jlong299, 2022-05-11:
It is said in spec that load whole register has full set of EEW variants as a hint to aid internally data rearrangement. Why doesn't the store whole register need the hint EEW in the case of internal data rearrangement?
Thanks.

jlong299, 2022-05-12:
Thanks a lot.  What if the register 0-1 has EEW of 16 and register 2-3 has EEW of 32, and the store whole register wants to put register 0-3 into memory? Does that mean I need extra hardware to record the vtype of all registers?

aswaterman, 2022-05-12:
It's never necessary to use extra hardware to record the vtype of each register.  See, for example, the Spike implementation of RVV, which does no such thing.

jlong299, 2022-05-12:
Can one store whole register instruction operate on multiple registers with different EEW settings?

nick-knight, 2022-05-12:
Yes, software can do this.

jlong299, 2022-05-13:
Got it. Thank you so much.

jlong299, 2022-06-21:
So if vstart > 0, the number of prestart elements of Store Whole Register instruction should be calculated based on EEW = 8 ?

aswaterman, 2022-06-21:
That's right.

closed 2022-07-03 by jlong299