riscvarchive/riscv-v-spec #805: Why do you consider to use LMUL to specify how many registers used in a vector instruction?

Zissi-Lei, 2022-06-13:
I think programmers shouldn't consider how many vector registers used by a vector instruction. Instead, we can just use vl to specify how many elements we want to process and let the hardware to count how many registers the instruction occupies according to vl，rather than exposing it to programmers. Otherwise, we should be aware of the LMUL even in intrinsics. I want to know why you consider to add LMUL to RVV, thanks for your time!

eopXD, 2022-06-16:
> Instead, we can just use vl to specify how many elements we want to process and let the hardware to count how many registers the instruction occupies according to vl.

Why do you want to limit yourself to process XXX elements per loop (or per instruction). For performance you should be programming in VLA (Vector Length Agnostic) style, basically what the intrinsic proposed. In that sense you program is scalable when you have a wider VLEN.

On the other hand, for performance you will also want to be aware of the register pressure, which directs to awareness of LMUL.

brucehoult, 2022-07-12:
If you request AVL = 1_000_000 (as is perfectly acceptable) you don't want the hardware to allocate all 32 V registers to one variable.

LMUL is used to tell the hardware how many variables you need in your loop (maximum 32/LMUL).

kasanovic, 2022-07-16:
Bruce explained this well.   A single instruction does not provide enough context for how many vector register groups are desired to hold temporary values.

closed 2022-07-16 by kasanovic