riscvarchive/riscv-v-spec #806: Question about element order for vfredmin/vfredmax

jinyinghan324, 2022-06-17:
In RISCV spec, it is said that for FMIN and FMAX, if at least one input is a signaling NaN, the result is the canonical NaN. If one operand is a quiet NaN and the other is not a NaN, the result is the non-NaN operand.
So if there is an element is a signaling NaN for vfredmin/vfredmax, different element order could result different outcome.
For example, assume vs2[0] = 0.1, vs2[1] = sNaN, vs1[0] = 0.2.
fmin(fmin(vs2[0],vs2[1]),vs1[0]) = 0.2
fmin(fmin(vs2[0],vs1[0]),v2[1]) = canonical NaN
fmin(fmin(vs1[0],vs2[1]),vs2[0]) = 0.1
In riscv-v -pec, element order is not provided as vfredosum. And it is not said that the outcome could be different due to different element order.
The designer could implement vfredmin/vfredmax in two ways.
1、In a fixed element order under given vtype and vl.    Different CPUs could have different outcome.
2、If any active element is sNaN, the outcome is canonical NaN.
Which way is expected?

aswaterman, 2022-06-17:
> In RISCV spec, it is said that for FMIN and FMAX, if at least one input is a signaling NaN, the result is the canonical NaN.

It turns out this isn't true for the fmin/fmax instructions.  The spec says, "If only one operand is a NaN, the result is the non-NaN operand. Signaling NaN inputs set the invalid operation exception flag, even when the result is not NaN."

So, for example, fmin(0.0, sNaN) = 0.0 (and the invalid exception flag is set).  The operator is associative, and so there is no concern about the element order.

jinyinghan324, 2022-06-20:
> It turns out this isn't true for the fmin/fmax instructions. The spec says, "If only one operand is a NaN, the result is the non-NaN operand. Signaling NaN inputs set the invalid operation exception [...]

In 《The RISC-V Instruction Set Manual Version 2.2》 Chapter 8.3 NaN Generation and Propagation. 
"For FMIN and FMAX, if at least one input is a signaling NaN, or if both inputs are quiet NaNs, the result is the canonical NaN. If one operand is a quiet NaN and the other is not a NaN, the result is the non-NaN operand."

aswaterman, 2022-06-20:
You are reading a version of the spec that predates ratification.  In 2017, the definition of these instructions was changed.  The spec was ultimately ratified in 2019 using the new definitions.

The version of the spec available at the RISC-V web site (https://riscv.org/technical/specifications/) reflects the correct definitions.

closed 2022-06-20 by aswaterman