riscvarchive/riscv-v-spec #808: What is the intended function of 12.2. Vector Single-Width Averaging Add and Subtract

kdockser, 2022-07-05:
The text in the Vector spec says that these instructions operate on unsigned and signed integers.
For example, it says that vasubu.vv returns the `Averaging subtract of unsigned integers`. 
Taken at face value: the inputs are unsigned, but the difference is necessarily signed. A negative difference _should_ remain negative after being averaged (i.e., shifted to the right by 1). This appears to be what Spike is doing.

However, the comment next to the instruction shows that the operation performed is `# roundoff_unsigned(vs2[i] - vs1[i], 1)`
Which is subsequently defined as `roundoff_unsigned(v, d) = (unsigned(v) >> d) + r`. We are no longer operating on the difference of unsigned integers but rather on a difference of integers that is recast as an unsigned value. Instead of a rounded negative average, we have a large positive value as a result. 

So, which is correct: the english prose in the spec (and Spike) or the "code" given as a comment in the spec?

nick-knight, 2022-07-06:
>  Instead of a rounded negative average, we have a large positive value as a result.

If you interpreted the bit-pattern of that large positive value as an encoding of a two's complement integer, wouldn't it be the same as the rounded negative average?

nick-knight, 2022-07-06:
Thanks for the explanation; I agree the pseudocode appears to be in conflict with the prose.

Rereading this, I also think the pseudocode is ambiguous regarding the data widths involved: the prose implies avoiding overflow in computing the intermediate sum or difference.

aswaterman, 2022-07-06:
Sounds like someone should propose an improvement to the pseudocode.