riscvarchive/riscv-v-spec #811: Concerns about surfacing mask bits as the low bits of v0

cmuratori, 2022-07-11:
While I can appreciate why RISC-V V 1.0 can't select among mask registers per instruction, I am troubled by the rationale (or lack thereof) behind the odd dual-use v0 scheme chosen in lieu of a dedicated mask register.

Using the low-order bits of v0 as the mask for vector instructions creates a significant burden on chip designers going forward. From now until the end of RISC-V V's useful lifetime, every chip will have to support surfacing the low-order bits of v0 as both an all-lanes mask and as the bottom few lanes of a vector. This is a very unusual decision, and makes very little apparent sense: mask bits will want to be near the lanes for which they are used, whereas the low-order bits of v0 will want to be near the low lanes. Why conflate the two?

With the current design, every chip designer trying to design a high-performance chip seems left with the unfortunate task of shadowing these bits somehow. They will be forced to implement various hacks to ensure that if the instruction stream uses v0 as v0, the bits will be in one place on the chip, whereas if it uses v0 as the mask, the bits will be in another place entirely. This seems strictly worse compared to having a single dedicated mask register, especially since RISC-V _already introduces_ several mask-specific instructions, so it cannot be said that the goal was to avoid having any special instructions just for masks.

Furthermore, it is unlikely that future chips will want to restrict themselves to a single mask register. Were RISC-V V v1.0 written in terms of a single dedicated mask, adding a RISC-V extension in the future that increases the number of masks would be a trivial extension and would leverage the existing design and programming knowledge. Instead, such an extension will have to be a complete rewrite of how masking works, since now there will be two totally different ideas of masking: one legacy version for the low bits of v0, that all chips will have to support for backwards compatibility, and then a new version which treats mask registers as something separate.

Looking through the materials I was able to find about the evolution of RISC-V V, it is hard to understand the reasoning behind the dual-use-v0 design. It is difficult to understand why a scheme like this was selected, especially in a brand-new spec that has no existing backwards compatibility constraints. I understand why RISC-V V instructions could not select a mask register per instruction, because there weren't enough bits in the instruction encoding. But I do not understand the rationale behind not having dedicated mask registers at all (while keeping the constraint that "m0" always be the mask used, instead of v0, since the instructions don't have enough bits to select), and barring that, I especially don't understand the rationale behind choosing the dual-use v0 scheme instead of just having one dedicated mask register.

Am I the only person concerned about this?  This seems like a serious issue, and I would like to understand what is going on here.

\- Casey

brucehoult, 2022-07-11:
v1.0 is ratified. If you wanted to raise this question, the appropriate time would have been a year or more ago. There is absolutely zero chance of it changing in future as RISC-V makes an absolute backwards compatibility guarantee.

Having separate mask registers as some, but by no means all, other ISAs do was explicitly considered by the working group, and rejected. Some members were for it, but not sufficient to do it. The implications of the decision were, I believe, well understood.

You are probably right that many complex high performance implementations will use a shadow mask register. That's hardly a big deal. Your point -- which you expounded on at length in your twitch stream [1] -- that this can imply moving bits between the shadow mask register and v0 is as far as I can tell not as serious as you seem to think. It would be extremely uncommon to switch uses of the same bits. A given set of bits is either a vector or a mask.

Future versions of RVV that use larger encodings that allow direct use of multiple mask registers will almost certainly not suddenly use a set of new mask registers. They will continue to use normal vector registers. Most algorithms don't need 32 distinct live vector variables and use LMUL of usually at least 4. If you are using v0 for masking then with LMUL=4 you already have v1, v2, and v3 available for use holding masks at no extra cost. This is enough for representing the control flow structures of most loops, at least according to Steve Wallach [2] in one email thread you referred to in your twitch stream.

In my opinion, a key strength of the RVV 1.0 design is that it enables flexible trade-offs that maximise the productive use of as much of the large and expensive vector register file as possible, over a wide range of types of code.

You complain that using v0 as a mask register wastes most of the bits in it. First of all, that's simply not true for small element sizes and/or large LMUL. In particular when SEW is 8 bits and LMUL=8, every single bit in v0 is used as a mask for some byte in the vectors being operated on.

Having a large dedicated mask register set (8 in AVX512, each 64 bits wide with the AVX-512BW extension allowing 8 and 16 bit elements; 16 in SVE, each 1/8th of the vector register size so 16-256 bits) means that in code that doesn't use masking, or uses few masks, there is a lot of wasted register space.

RVV lets you use the entire register set space for data when you're not using masking (which is very often, since RVV has a vector length register and so unlike SVE doesn't need to use masks just to control the scalable vector size), and flexibly partition it into mask space and data space when you do want masking.

Similarly, LMUL lets you use most or all of that expensive vector register space for data even on algorithms that require only a handful of live vector variables. You can go to as few as 4 variables, each of 8x the nominal vector register length. Or 3 variables plus 4 masks.

And you STILL can use the one loop body to process short vectors or the tail of a long vector.

On SIMD if you need fewer logical variables than you have registers then you can unroll your loop by hand and process 2 or 4 times as much data in each loop iteration, to use up all your registers. But this just makes the contrast between the main loop and the shot vector/tail handling even larger -- or requires even more versions of the loop body.

As you said on your twitch stream, this is the only decision in the entire design that you have serious concerns about. "It doesn't feel like there are all these weird mistakes everywhere". "If you just changed this one thing, the spec would be fine". Those are direct quotes from you.

There have been literally thousands of individual decisions made in the design of RVV, with numerous trade-offs all over the place. I'm sure there is not one member of the working group who got their way on every single design choice. That's just a part of designing something that will be used to scale the whole range from microcontrollers to supercomputers. Something that, I note, SVE hasn't done otherwise they wouldn't need MVE.

This is all of course entirely my personal opinion as a nobody. I'm an individual member of RISC-V International (as you could be yourself if you wanted to influence things), not affiliated with any company, but you will find my name in the contributor's list for RVV as well as the B extension and the RISC-V ISA Manual itself so I'm maybe not a complete ignoramus. But I perhaps have the freedom to be less diplomatic than people representing companies here.

[1] https://www.twitch.tv/videos/1508466652

[2] famous designer of high performance vector machines https://en.wikipedia.org/wiki/Steve_Wallach

cmuratori, 2022-07-11:
If there's no chance of changing this (ever), then there's no need to discuss it, but just so I make sure I understand the reasoning:

If I understand your explanation correctly, the background assumption for this design is that RISC-V V is _never_ intended to have mask registers at all, even in the future as an optional extension, so this is the design for the instruction set assuming that there will always be 32 total shared register names that serve the dual purpose?

- Casey

brucehoult, 2022-07-11:
Register names are not limited to 32 in future instruction encodings any more than mask registers are limited to 1.

I suppose I can't say that there can never be a decision to add mask registers in a different namespace, but as you point out that would be extremely inconvenient as it would make the current mask instructions redundant and baggage that needs to be maintained forever just in case someone uses them.  The logical conclusion from that is that if it was intended to have a lot of mask registers in a separate namespace in future then the current 1 mask register would have been independent of v0 from the start. But it is not. And not because no one thought of that possibility.

Your assumption that a decision different than you would have made are the result of inexperience ("this is what *I* would have done, before I did any vector programming") or not considering alternatives and their ramifications is, in my opinion, unfounded.

cmuratori, 2022-07-11:
I guess I'm not clear on where someone would get extensive real-world experience with a mask-registerless-low-bits-as-mask design given that RISC-V V is so new and I don't know any other widely-adopted ISA that works this way? If you just mean SIMD programming experience in general, I would wager I have as much experience as most writing SIMD workload code, since that's been much of my job, much of the time, for several years.

One of my current jobs is evaluating the RISC-V V ISA to see if it is suitable for certain high-performance workloads. In order to do this, I have to look at things like how the masking is going to work, and you have to analyze loops and predict what the pain points are. At least in the preliminary stages, I have a hard time agreeing with several of the things you suggested in your earlier reply as reasons not to be concerned. For example, "most algorithms don't need 32 distinct live vector variables" is fine to say if you are talking about simple scalar loops in general-purpose code. It is less fine, on the other hand, when you start to look at SIMD loops which do vector gather, since these gathers can be quite slow and want to be launched very early, which means you either need a lot more registers to handle the pre-issue, or you need to insert spurious spills.

So when you're evaluating something like this, it's pretty important to understand the rationale behind what was done, so you understand what the path forward is supposed to be. It makes a big difference whether the answer is "we're starting without mask registers but we plan on adding them" vs. "we're never having mask registers", since one assumes there's never going to be more than 32 vector registers.

If you're picking an ISA, you're picking a long-term strategy, and if that strategy is definitely not going in a direction you're comfortable with, that's the end of the analysis and you go with something else. Either way, I do think it was useful to hear your response, because I guess I would say I'm considerably _more_ worried about this design now than I was, since previously I was under the assumption that mask registers were going to be introduced later, whereas I guess the idea is that RISC-V V is just going to be 32 registers, full stop. So my concern went from "this design decision seems weird and might negatively impact the design of the chip" to "now we should do this evaluation assuming we will never get mask registers in future revisions".

It's still a bit early, so it may be that after really working on RISC-V V-specific code for several months and running full simulations, it turns out that "32 registers for vectors and masks is enough forever". That would be great. But I can't say I'm optimistic about that, and in general it is also a little bit more scary choosing RISC-V V if it will always be idiosyncratic in this way, given that code for the other major architectures all have 32 registers plus the masks, rather than 32 registers minus the masks.

\- Casey

nick-knight, 2022-07-11:
Tagging related discussion: https://github.com/riscv/riscv-v-spec/issues/617.

brucehoult, 2022-07-12:
> one assumes there's never going to be more than 32 vector registers.

> I guess the idea is that RISC-V V is just going to be 32 registers, full stop

> it turns out that "32 registers for vectors and masks is enough forever"

You apparently missed the very first line in my message: "Register names are not limited to 32 in future instruction encodings any more than mask registers are limited to 1"

A future version of RVV with longer instruction encodings e.g. 64 bits will have room for register number fields larger than 5 bits. They could be 6 or even 8 bits, with correspondingly more registers, in exactly the same way as the base RISC-V ISA has 5 bit register fields while most C extension instructions have 3 bit register fields. The current 32 bit RVV opcodes can map as special cases of the 64 bit opcodes, and be automatically picked by the assembler when possible, just as C extension instructions are now.

This is much more straightforward than adding dedicated mask registers later.

You could also retain 32 bit opcodes but do a REX/VEX prefix trick to expand the register set. Vomit.

> I guess I'm not clear on where someone would get extensive real-world experience with a mask-registerless-low-bits-as-mask design given that RISC-V V is so new and I don't know any other widely-adopted ISA that works this way?

Are you suggesting it is too dangerous for anyone to ever do anything new or different to everyone else?

> If you just mean SIMD programming experience in general, I would wager I have as much experience as most writing SIMD workload code, since that's been much of my job, much of the time, for several years.

SIMD experience is maybe not as transferrable to vector as you might think. But for what its worth I've been doing SIMD in assembly language since the PowerPC G4 in 1999.

cmuratori, 2022-07-12:
> You apparently missed the very first line in my message:
> "Register names are not limited to 32 in future instruction 
> encodings any more than mask registers are limited to 1"

I did not miss it, I didn't think it was particularly relevant.

First, to be clear, because "mask registers are limited to 1" might suggest to someone that I'm talking about mask selection: I am not talking here about the ability to _select_ mask registers in the instruction. I'm talking about the ability to have _dedicated_ mask registers instead of overlaying that functionality onto the vector registers. Obviously "mask registers" of the kind I'm talking about aren't "limited to 1" in the current RISC-V V spec, they're limited to zero, because there aren't any. Mask _selection_ is limited to "v0 or nothing", but that is not something I flagged as a high-priority concern. Maybe I should be more concerned about it, but I didn't feel like the extra register renames required by a dedicated mask source were that worrisome.

Second, sure, I believe you that someone might propose a 64-bit instruction encoding option for RISC-V V that extends the vector register names to 64 without causing other significant problems. But that would be a completely new proposal that doesn't currently exist, and I have no way of making any guesses about the qualities of this hypothetical change. It certainly seems like a bad trade-off at first blush since proper mask registers would have kept the encoding at 32 bits, but, maybe having 64 vector registers would turn out to be an unexpected sweet spot for other reasons. I've seen architectures with 128 vector registers (spoiler alert for later in the post!), so, you know, never say never. But either way, I don't know what I do with such hypotheticals _today_.

Third, just to restate: my operating assumption had been that mask registers were a plausible near-term RISC-V extension, so having a full 32-register name set was likely with no major changes. Based on things that were mentioned during discussions about RISC-V elsewhere, I got the impression perhaps it had been left out of RISC-V V 1.0 because it was going to be "tiered" such that mask registers would not be required on lower-end hardware. This was also why I was concerned about the odd v0 mask scheme that was the gravamen of my original post, because I could not see how this was going to make sense once mask registers were added.

However, it sounds like actually I was very mistaken, and mask registers are not on the horizon. That's actually great to know. I can just cross that possibility off the list, and I can assume the 32-register v0 design can just be assumed to be "it" for the foreseeable future.

Maybe that means we shouldn't have been looking at RISC-V in the first place, and I'm just wasting everyone's time with this post. That certainly seems to be the case. Unfortunately, I had no way of advising against that ahead of time, because newcomers to the ISA basically just get the spec and the minutes to work through, and you don't necessarily know what the trajectory of the project is. Since things like #617 are marked "resolve after v1.0" that sort-of reinforced the impression that the mask situation was not considered ideal, and would perhaps be revised. Knowing that's not the case is, as I said, actually very helpful.

> Are you suggesting it is too dangerous for anyone to ever do 
> anything new or different to everyone else?

No, I am responding to your statement that I must not know what I'm doing because I didn't have enough experience. My point was, nobody has enough experience with RISC-V V to make a definitive statement about anything at this point. It will be some time before there is enough practical work to see how everything turns out. Similarly, nobody has enough experience with vector SIMD in general for many relevant workloads, because all consumer high-performance chips have traditionally been packed SIMD, so that's what has the most proofing for these workloads. So the only qualifications somebody can have for evaluating RISC-V V for modern workloads is if they have extensive _packed_ SIMD experience, which I have.

But to answer the question directly, trying something new and different with RISC-V might be a great idea. More power to you if that's what you're into! But when evaluating ISA choices from the perspective of someone picking among the available options, there is no particular benefit to "new and different". It only matters if the performance evaluates well when simulated, and that you can reasonably forecast that the spec will continue in a direction that works for your use case. "New and different" isn't relevant to the customer unless it actually turns out to be better.

> But for what its worth I've been doing SIMD in assembly language 
> since the PowerPC G4 in 1999.

I guess I'm not sure how that is relevant. Although you called into question my experience, I did not call into question yours, and for the avoidance of doubt I can assure you that it does not make a difference to me either way. The reality of the design doesn't change based on who is talking about it.

Tangentially, what I remember most about the PowerPC when it was prevalent was that it forced me to support big-endian everywhere which always made me grumpy. I never had to do the PowerPC-specific optimizations in the G4 era because Mac (and GameCube) ports were done by someone else, so the first time I touched PowerPC SIMD specifically was actually on the 360, which of course was quite a bit different than the G4.

\- Casey

brucehoult, 2022-07-12:
> Maybe that means we shouldn't have been looking at RISC-V in the first place, and I'm just wasting everyone's time with this post. That certainly seems to be the case. Unfortunately, I had no way of [...]

You shouldn't try to intuit a "trajectory", you should look at what is in the ratified spec. As we are still likely six months to a year from having any RVV 1.0 hardware, we are probably five years from having some updated version.

> No, I am responding to your statement that I must not know what I'm doing because I didn't have enough experience.

> Although you called into question my experience

I made no such statement, sir.

cmuratori, 2022-07-12:
> You shouldn't try to intuit a "trajectory", you should look at what is in the ratified spec. 

That is exactly what I did do, and have concerns. That's why I posted the issue.

> As we are still likely six months to a year from having any RVV 1.0 hardware, we are 
> probably five years from having some updated version.

Certainly. I am specifically talking about evaluating that hardware for the near-term, and doing due diligence on what is likely to happen in the five-year version. That is the point of this inquiry.

Deciding whether to go with a RISC-V core design means you evaluate a particular implementation of the current spec to see if it suits your immediate needs, _and_ you try to determine whether the spec will evolve in the direction you need it to go down the road. Asking questions like why a particular decision was made a particular way is very important for determining whether you trust the future of a particular design aligns with the future of your platform. Obviously you can never know for sure, but some answers will give you high confidence, and others not.

\- Casey

aswaterman, 2022-07-12:
Casey's OP highlights an entirely valid concern.  And although ratified RISC-V ISAs aren't mutable, there's nothing wrong with having a spirited discussion about one of the most contentious aspects of the V extension's design, so long as it stays respectful.

One of the common misconceptions has been that masks need only be a few bits wide, and that's not the case in RVV: since we support masking of LMUL=8 SEW=8 vectors, masks can be as large as VLEN bits.  This observation largely motivates the ISA's masking scheme.

I want to highlight Bruce's point that we didn't arrive at this scheme lightly.  We carefully considered a variety of alternatives before settling upon this one as what we perceived to be the least-problematic alternative.  Among some of those considerations were:
- Dynamically, most kernels don't use masks; dedicating state to masks pessimises the common case
- Architectural simplicity favors less new architectural state, since new instructions would be needed to access it
- Context-switching overhead favors less total architectural state
- Low-cost designs favor less total architectural state
- Since mask registers need to be VLEN bits wide, an alternative scheme with additional mask registers would be expensive, or would require a register-grouping scheme similar to the current one
- The mask datapath problem described in the OP is real, but widening arithmetic operations present the same problem.  Microarchitectures that find the former problematic will find the same of the latter, and probably sooner, by at least a generation [citation needed].

Some of those points are perhaps arguable; I'm just trying to give a flavor of our rationale.

Having worked on RVV microarchitectures that run the complexity gamut, I've both praised and cursed this architectural feature.  I'm confident what we ratified is near or along the Pareto frontier, even though it might frustrate.

cmuratori, 2022-07-12:
Thank you Andrew that is a very illuminating list, and very helpful!

If you wouldn't mind elaborating on a few points I can go ahead and close the issue:

1. How was it determined that "most kernels don't use masks"? This may be where I was most confused in my original understanding of RISC-V V, because for the workloads with which I am familiar (all of graphics, all of OpenCL/CUDA/etc.), basically everything is masked. It is extremely rare to find something that does not need masking (due to if's and conditional loops). It sounds like RISC-V V is thinking about something very different than these sorts of workloads as standard, and it would be helpful to know what it is more broadly.
2. Regarding "an alternative scheme with additional mask registers would be expensive", do you mean specifically with respect to low-cost hardware only? I assume that there is no real expense saved on the high-cost side, since once you have to implement some kind of shadowing or optimization scheme, you are effectively already paying that expense, you're just not getting the additional register names in return, no?
3. Does widening actually present the same problem, or are they actually two different problems? When you widen something, that is typically an actual operation, so the widening occurs in the context of the execution part of the pipeline. "Automatic" mask widening as it is in RISC-V occurs in a different place, at the point where the mask is supplied to start the operation in the first place. This is sort-of a separate issue, because it has to do with whether you define your masks as "automatically" widening and that decision could be made independently of the rest of this, but it need not have been made the way it was in RISC-V V so it does seem relevant to ask the rationale in the context of this point.

The other points you listed make sense to me as they are, since they all follow from the first one (eg., if the design space of RISC-V V doesn't consider predicated operation to be common, then obviously you would not want to pay extra anywhere for providing it).

Finally, for us specifically I should ask: are we looking at the wrong RISC-V spec? Is there a different spec planned somewhere that is going to be for OpenCL/CUDA/etc. processing that would be making these tradeoffs differently (eg., be based on the expectation that most things need masks, that favor high-cost over low-cost designs, etc.)?

Thanks,
\- Casey

ratchetfreak, 2022-07-12:
> Dynamically, most kernels don't use masks; dedicating state to masks pessimises the common case

which kernels did you look at? and in which instruction sets were they? and did they have masked instructions?

If the last question there was no then it's very likely that psuedo masks were used that tend to hide in other operations (like muls with some values 0 or sign-extension bit trickery etc. ) selection bias against masks is strong given the history of the various simd specs where masking was similarly undervalued during development.

brucehoult, 2022-07-12:
Thanks indeed to Andrew for taking the time to comment. All of his points were previously made by me in my replies to Casey but it certainly doesn't hurt to have Andrew confirm them.

RISC-V was indeed designed to be good for OpenCL/CUDA workloads. If you look at RISC-V co-inventor and SiFIve co-founder Yunsup's Lee's 2016 PhD thesis [1] you will find about half of it dedicated to the advantages of scalarising SIMT code to run on a vector ISA rather than vectorising C code, complete with details on how to go about the transformation, including dealing with divergence and convergence.

However that is not the *only* application area for RISC-V V extension. In microcontrollers and "edge" applications generally, the main workloads are likely to be media processing and ML, neither of which tends to use masking, as far as I'm aware.

> Is there a different spec planned somewhere that [...] favor high-cost over low-cost designs, etc

No. The one spec covers different kinds of workloads and both low cost and high performance designs.

As I said to you in my first reply in this thread:

"There are many places in the design of RISC-V intended to make simple implementations simple, and high performance implementations possible, at possibly a little extra complexity in the already-complex implementations that can best afford it."

[1] https://www2.eecs.berkeley.edu/Pubs/TechRpts/2016/EECS-2016-117.html

brucehoult, 2022-07-12:
> which kernels did you look at? and in which instruction sets were they? and did they have masked instructions?

Perhaps you are not aware that RISC-V itself came out of a vector processing research group at Berkeley, where it was intended (before the rest of the world expressed a demand for it) to simply be the scalar control unit for vector processors. All the early people associated with RISC-V, and subsequently working on the V extension, are familiar with a wide range of vector hardware (not just the limited SIMD found from Altivec to MMX to SSE to AVX to Neon) and vector algorithms.

It's not like RISC-V already existed for a while and then one day people discovered that there was a need for a vector extension so ... quick, we don't know much about it but let's bodge something up.

cmuratori, 2022-07-12:
> RISC-V was indeed designed to be good for OpenCL/CUDA workloads. 
> If you look at RISC-V co-inventor and SiFIve co-founder Yunsup's Lee's 2016 PhD thesis 

I do not understand how this is relevant to the discussion of mask registers. Can you cite which part of that thesis you think explains why predication is "not necessary for most kernels"? The thesis itself hypothesizes a programming model with 16 dedicated predicate registers. If I may quote from Chapter 6 where the architecture is explained:

`Chapter 5 shows that simple vector predication in conjunction with consensual branches is able to efficiently handle complex control flow (if-then-else statements, loops, function calls including virtual calls) found in parallel programming languages. The Hwacha vector-fetch architecture stores predicates in vector predicate registers (vp0–15) and allows all vector instructions (including both vector computational instructions and vector memory access instructions) to be predicated.`

> "There are many places in the design of RISC-V intended to make simple 
> implementations simple, and high performance implementations possible, 
> at possibly a little extra complexity in the already-complex implementations 
> that can best afford it."

I think we are all aware of that sentiment at this point. But having a goal of "possibly a little extra complexity" does not mean that that goal was actually achieved. It may be that in making decisions to favor low-cost implementations, the spec actually did make things too difficult for high-performance implementations down the road.

\- Casey

brucehoult, 2022-07-12:
> I do not understand how this is relevant to the discussion of mask registers. Can you cite which part of that thesis you think explains why predication is "not necessary for most kernels"?

Apparently you misunderstand me.

I bring up Yunsup's work to point out that the needs of translating SIMT code to vectors -- which includes masks, inverting masks, ANDing and ORing masks, efficiently detecting all 0 and all 1 masks, and the RVV iota instruction (for get_local_id(), get_global_id()) -- has always been an explicit part of the reason for RISC-V Vectors and considered in the design.

That is a large part of the expected workload that needs masking. As is auto-vectorising random C code.

The kernels that predication is not necessary for includes OTHER stuff. Sound and video processing. ML, I believe. Lots of stuff in BLAS and LAPACK and the like (dense linear algebra).

ratchetfreak, 2022-07-12:
So the target for risc-v architecture is not going to be general purpose procedural code, but instead the "OTHER" stuff?

Also I'm fairly sure that video processing is going to need masking (given that term came from doing image manipulation back in ye olden analogue days).

aswaterman, 2022-07-12:
> * How was it determined that "most kernels don't use masks"? This may be where I was most confused in my original understanding of RISC-V V, because for the workloads with which I am familiar (all [...]

Well, nothing's quantifiable without a workload, so that statement of mine was inherently brash.  But the dynamic frequency of e.g. memcpy and memset does make the point, to some extent.

More relevant to this particular discussion, and perhaps partially addressing @ratchetfreak's point, Yunsup's dissertation highlights that much _apparent_ divergence in CUDA-style code is synthetic, not actual.  One of the most common control-flow idioms in that domain is of the form of `if (thread_id == 0) { ... }`, which is fundamentally _scalar_ control flow--it's code that should not be running on a vector processor at all.  (Yunsup's thesis addresses a more interesting and general problem than this, and he presents a compiler framework for dealing with it.  But the running theme is that a bit more compiler sophistication significantly reduces the pressure on vector control-flow mechanisms.)

> * Regarding "an alternative scheme with additional mask registers would be expensive", do you mean specifically with respect to low-cost hardware only?

The cost is more pronounced at the low end, for sure, but additional architectural state never comes for free.

> Does widening actually present the same problem, or are they actually two different problems? When you widen something, that is typically an actual operation, so the widening occurs in the context of the execution part of the pipeline.

The problems are differently motivated, but they have a common solution.  Naturally, the choice to optimize either is a microarchitectural decision, but wide designs that want to efficiently support widening operations will need these mechanisms anyway.  (And for narrow designs, neither concern really matters that much.)

cmuratori, 2022-07-12:
OK thanks, I think I've got the gist of it.

\- Casey

closed 2022-07-12 by cmuratori

freemin7, 2023-07-12:
If one year ago it was too late, it is double too late now. So i am just adding this for context. I haven't looked deeply into the RISC-V Vector instruction set (but am getting a Milk-V Duo soon with versio 0.7.1 to have a reason to do that) however i find it deeply discouraging what i read here and believe to understand it.

For reference i am currently using a NEC VectorEngine. A proprietary ISA which has 64bit wide instructions. 64 64 bit wide scalar register. 64 16348 bit wide vectors and 16 256 bit wide vector masks spanning the entire maximum vector length of 256 64 bit words in a vector.

Having an optimizing compiler for this architecture i can absolutely not support the assertion that vector masks often go unusused.

They make certain applications vectorizable at all. That the thread_id==0 check is the only example brought up in the thread makes me sad.

I have some applications where i make heavy use of the mask registers to build a per vector element state machine.

rvvinfo, 2023-07-16:
> getting a Milk-V Duo soon with versio 0.7.1

Cool. RVV 0.7.1 has the mask bits laid out as Casey wants them to be.

> For reference i am currently using a NEC VectorEngine. A proprietary ISA which has 64bit wide instructions. 64 64 bit wide scalar register. 64 16348 bit wide vectors and 16 256 bit wide vector masks spanning the entire maximum vector length of 256 64 bit words in a vector.

Sounds like a very impressive machine! Other than not being able to mask 8, 16, or 32 bit data items, apparently. And it has twice as many vector and scalar registers as mainstream Aarch64, RISC-V, and x86. RVV 1.0 supports the same or larger total vector register size, but with 32 registers of 32768 (or even 65536) bits. SVE and AVX512 don't, being limited to 2048 and 512 bit registers respectively.

Probably costs a bit more than that $9 Duo.

> Having an optimizing compiler for this architecture i can absolutely not support the assertion that vector masks often go unusused.

They DO often go unused. And they often go used too.

Many widespread and successful SIMD ISAs don't directly support masking at all, they make you use additional "select" or "and" instructions. That includes NEON, SSE, AVX.

SSE, AVX, and 32 bit NEON only have 8, 16, 16 vector registers, so taking away some of those for use as pseudo-masks is very painful. RVV and 64 bit NEON have 32 registers, so you are much more likely to be able to spare some for use as masks.

> They [masks] make certain applications vectorizable at all. 

Of course. No one denies that. That's why RVV makes almost all instructions optionally maskable.

> However that only V0 can be used to predicate instructions makes running two interleaved instances of a latency bound program which uses less than half of all registers require a lot more shuffling and a lot of op fusion. This will also negatively impact the count of useful instructions in out of order reordering window.

Nothing is for free. We only have 32 bit instructions at present. If each instruction included the source register for its mask then something else probably more important would have to be chopped. For example, vector instructions would have to be "dst = dst op src" rather than "dst = src1 op src2" as they are now. That would add a lot of move instructions to many algorithms, with heavy penalty on simple non-renamed implementations such as your $9 Duo.

If you have to insert an extra "move mask to v0" instruction before every (which is unlikely) RVV instruction in your program then you'll be using effectively 64 instruction bits per operation, same as your NEC VectorEngine. With still fewer addressable registers, certainly.

A future RISC-V RVV 2.0 with 64 bit instructions can increase the number of vector registers to 64, 128, 256 ... and also allow the selection of a different mask register directly in each instruction.

Feel free to join in helping with the design of that.

freemin7, 2023-07-16:
> Sounds like a very impressive machine! 

I am currently revamping my vector host hoping that a first gen threadripper will work. Noone knows whether that works or not. I will also be getting a second VectorEngine in the same machine and after no horrible experiences running it as a shared server stealthy. I plan to open it up to people more liberally. 

Now a few architectural comments

> Other than not being able to mask 8, 16, or 32 bit data items, apparently. 

32 bit data is supported for integers and floating point numbers with full throughput (twice the 64 bit throughput). The newest vector engine generation added fully predicated 16 bit float/bfloat but the throughput is 32 bit throughput so you just get even higher bytes/flop. A VectorEngine comes with six stacks of HBM and draws below/around 300 Watts.

The server versions of the 10E draw have the same memory bandwidth as intels new sapphire rapids. While having one 8 PIN EPS12V connector

> And it has twice as many vector and scalar registers as mainstream Aarch64, RISC-V, and x86.

Yes it's purpose is to have twiddle thumbs as little as possible while waiting for data and to have a small core count. The model i have has 8 cores at 1400 MHz and it's double precision throughput is the ballpark of two 4090s while drawing less power then one in GEMM for which it was to much memory bandwidth.

>  RVV 1.0 supports the same or larger total vector register size, but with 32 registers of 32768 (or even 65536) bits. SVE and AVX512 don't, being limited to 2048 and 512 bit registers respectively.

Well. For the Aurora architecture (all vector engine the MVL is 256 (meaning 256*64 bits registers)). The architecture is MVL agnostic. MVL is a standalone 64 bit register meaning any integer is allowed by the ISA and the vector size can exceed the size addressable by a 64 bit address space by a factor of log2(64bit/8bit). Fun.

> If you have to insert an extra "move mask to v0" instruction before every (which is unlikely) RVV instruction in your program then you'll be using effectively 64 instruction bits per operation, same as your NEC VectorEngine. With still fewer addressable registers, certainly.

The cores on a NEC VectorEngine *are* out-of-order cores for both the scalar and the vector part. My concern is that this architecture could make reordering a lot harder might be unfounded. I think the ability to hide memory latency by reordering supposedly makes a big difference for the VE... I haven't benchmarked that claim yet but there is a way to turn reordering off of the VE.

Also i imagine that for particular iterative operations relying on popcnt(mask) dedicated masks are an advantage.

> Feel free to join in helping with the design of that.

I would very much like to be involved in this both by providing NDA free and uncomplicated access to a (or two) workstation tier NEC VectorEngines and involving myself in RVV. However i have no idea how to do the later.

rvvinfo, 2023-07-17:
>> Other than not being able to mask 8, 16, or 32 bit data items, apparently.
>
>32 bit data is supported for integers and floating point numbers with full throughput (twice the 64 bit throughput).

How does it do that when mask registers have only 256 bits?

16384/32 = 512.  And of course float16/bfloat vectors would need 1024 mask bits. If you have to gang together mask registers to address all elements then 16 mask registers won't go very far.  And if you don't, then you're going to be able to use only a 1/2 or 1/4 or 1/8 of each vector register, lowering throughput correspondingly.

From my calculations from the claimed TFLOPS, dividing by 8 CPUs, 1400 MHz, and 256 64 bit elements per vector register it appears that the 64 bit throughput is 3 IPC, implying not only three pipes but also full-width 16384 bit ALU.

freemin7, 2023-07-17:
> How does it do that (512 element ops) when mask registers have only 256 bits?
Uses two neighboring masks as one mask.

> From my calculations from the claimed TFLOPS, dividing by 8 CPUs, 1400 MHz, and 256 64 bit elements per vector register it appears that the 64 bit throughput is 3 IPC, implying not only three pipes but also full-width 16384 bit ALU.

My understand is that that they have 32*64 bit wide functional units. 3 of which can do FMA in one cycle on all elements.

rvvinfo, 2023-07-17:
> Uses two neighboring masks as one mask

So, for 8 bit elements, the 16 mask registers only give two masks, as each mask will require eight neighboring masks to be used as one mask.

Ouch.

Every design has trade-offs :-)

freemin7, 2023-07-17:
Even more interesting, they have a bunch of unused space in their ISA but they locked themselves out of continous access to larger masks because they did put some flags next to them. This allowed them to:
 - have less than 256 instructions
 - only 6 instruction formats across their scalar and vector ISA
 - have pretty trivial decoding logic (or so i would assume)
 - have pretty powerful instructions that involve scalar and vector registers
This instruction format limits them to 4 bits of mask count without using the last free
[image]
[image]
[image]
[image]
[image]
[image]

The NEC VectorEngine is designed for 64bit. So much so that non 64-bit alligned 32bit loads caused problems/pain before ve3 and that also left some warts on the ISA.

rvvinfo, 2023-07-17:
> However the underlying issues is that to fine vector masks in the limit become just a vector register.

Which is exactly what RISC-V does.

Byte-sized operations are an important use-case for RVV:

- accelerating <string.h> functions: memcpy, memcmp, memset, and the null-terminated str* functions. The vast majority of programs benefit from this instantly, simply by having an RVV-ified libc.
- lo res audio
- colour components of images/video
- ML

> Do you have a suggestion where to move further VectorEngine design discussion. It feels like we are getting off topic.

I agree.

VectorEngine is clearly aimed at particular workloads, and good for it. It puts a stake in the ground for what a $15k RISC-V HPC vector processor needs to match -- in overall performance, not necessarily in every detail of the design.

RVV is aimed at everything from microcontrollers, to your $9 1 GHz Linux-capable Duo, to supercomputers that perhaps will soon be in or beyond Fugaku class.

freemin7, 2023-07-17:
>> Do you have a suggestion where to move further VectorEngine design discussion. It feels like we are getting off topic.

> I agree.

Where would that be?

> Byte-sized operations are an important use-case for RVV:
> - accelerating <string.h> functions: memcpy, memcmp, memset, and the null-terminated str* functions. The vast majority of programs benefit from this instantly, simply by having an RVV-ified libc.

While i agree that it will be improving those use cases. However with [Intel DSA](https://cdrdv2-public.intel.com/759709/353216-data-streaming-accelerator-user-guide-2.pdf) it looks pretty clear doing it using vector/simd instructions is just a stop gap. That stop gap might help especially in embedded until those accelerators start being part of SoCs.

For str_* there is some pretty intersting prior art in the [TRONCHIP ISA](https://ieeexplore.ieee.org/document/378609) I don't have an english resource at hand on their string manipulation instructions but it might be worth looking into.

> - lo res audio

This is not credible. What would be credible is low precision measurements from embedded sensor. In embedded this makes much sense. However for embeeded DSPs have been popular. I don't think the CPU should try to eat part of the cake of a DSP.

> - colour components of images/video

Since those are stored as rgbrgb... that really tends to cause problem with vectorisation for most uses i can think of. In addition most rgb information is stored as rgbα so 8 bit instructions don't help much either as such operations tend to be memory bound so the byte masking isn't that bad. Although vectorisation can behave a bit nicer memory wise. If it is stored in seperate channel rrr... , bbb... , ggg... i see vector instructions doing much better. 

> - ML

I am curious what Jim Keller does with this. But i suspect a ReLu instruction will always beat a "FMA then make mask then zero based on mask" approach.

I am curious to see where this goes. Is there some way how i can get to know more of the people working on RVV?

rvvinfo, 2023-07-17:
> Since those are stored as rgbrgb... that really tends to cause problem with vectorisation

https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#sec-aos

    # Example 1
    # Memory structure holds packed RGB pixels (24-bit data structure, 8bpp)
    vsetvli a1, t0, e8, ta, ma
    vlseg3e8.v v8, (a0), vm
    # v8 holds the red pixels
    # v9 holds the green pixels
    # v10 holds the blue pixels