riscvarchive/riscv-v-spec #813: element groups are still broken in stripmining

jnk0le, 2022-07-16:
made an earlier issue in riscv/riscv-crypto/issues/148

according to paragraph 6.3 of RVV spec vl can be set to

> ceil(AVL / 2) ≤ vl ≤ VLMAX if AVL < (2 * VLMAX)

So if we want to stripmine 3 aes blocks with VLEN == 256, the uarch is allowed to split AVL equally to 192 bits (3*64bits) in both iterations due to better or more even lane utilization.

kasanovic, 2022-07-22:
Thanks for pointing out this case.
I added constraint in https://github.com/riscv/riscv-v-spec/commit/44c9b212c0928c6e8fefeb652ceb794810cd01d9 which addresses this issue.

topperc, 2022-07-22:
I think LLVM has assumed that a vsetvli with AVL>VLMAX for SEW=8 LMUL=1 and SEW=16 LMUL=2 would return the same VL. Is that no longer true?

aswaterman, 2022-07-22:
@topperc I agree this would be a backwards-incompatible change to a useful property. We need to retain the property that the VL setting is a function of only AVL and VLMAX.

kasanovic, 2022-07-22:
@topperc  Agreed - will work on a revised solution that preserves backwards-compatibility.

kasanovic, 2022-07-23:
A possible approach is to define a new variant of vsetvli that uses zimm[9:8] to hold a vegs[1:0] field that encodes an element-group size (EGS). This would use the same encoding as the old EDIV proposal (00=1, 01=2, 10=4, 11=8) but now this would only affect the operation of the vsetvl itself, not create any new state in vtype.  SEW and LMUL are treated as before to determine VLMAX.

The vsetvli instruction is now constrained to set vl to an integer multiple of EGS when AVL>VLMAX.  If this is not possible, vill is set. When the EGS field encodes 1 (vegs[1:0]=00), the instructions work as before (backwards compatible).

kasanovic, 2022-07-23:
If we constrain EGS to power-of-2 (POT), then we can simply constrain the load-balancing case across all SEW.  This needs no new instruction types, and is only minimally suboptimal for some niche design points assuming we only have POT EGS. I'll' update doc.

kasanovic, 2022-07-23:
Please review next revision in https://github.com/riscv/riscv-v-spec/commit/069c622ce015f21c6f24dd42b31582a72907dc33.

I believe this simplifies the discussion and provides backward compatibility.

aswaterman, 2022-07-24:
This revision satisfies my concerns.

closed 2023-02-23 by jnk0le