riscvarchive/riscv-v-spec #814: How to use RVV instructions to do the codegen for unordered comparison

zhongjuzhe, 2022-07-21:
Hi, I am optimizing auto-vectorization in GCC using RVV features. 
With -fno-trapping-math compile option, we will get unordered comparision.
For example
GCC IR:
mask__302.2354_418 = vect__143.2338_324 u>= vect__144.2342_357; (unordered ge between two vectors then generate the mask).

Can some one tell me the RVV codegen for this GCC IR?  
I tried the following coden:
```
vmfeq.vv        v1,va,va
vmfeq.vv        v2,vb,vb
vmand.mm    v0,v1,v2
vmfle.vv         v0,vb,va,v0.t (generate the ge result without rasing exception.)

vmnot v1,v1
vmnot v2,v2
vmor v1,v1,v2 (genarete unorder result)

vmor v0,v0,v1 (final result)
```
I do this with reference :https://stackoverflow.com/questions/8627331/what-does-ordered-unordered-comparison-mean.

I am not sure whether my understanding is correct. Thanks!

kito-cheng, 2022-07-21:
Backup and restore fflags, see scalar code gen here https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/riscv.md#L2337

C:
```
long
foo (double x, double y)
{
return __builtin_isgreaterequal (x, y);
}
```

ASM:
```
...
        frflags a1
        fle.d   a0, fa1, fa0
        fsflags a1
...
```

aswaterman, 2022-07-21:
Manipulating fflags is correct but is extremely slow for decoupled vector
machines. It might be worth trying to improve on the original proposal
rather than manipulating the flags.

nick-knight, 2022-07-21:
> However I get the wrong result.

Looks like `vmor v0,v0,v1` would assert the bits corresponding to NaN inputs-- don't you want to deassert those bits?

Regarding Kito's comment, it also depends on the (implementation-defined) treatment of sNaNs. The vector code in the original post will set NV if there are any sNaN elements in either input vector operand; Kito's proposal does not have this behavior. To `HONOR_SNANS`, GCC appends `feq.d zero, fa1, fa0` to the generated code [[ref](https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/riscv.md#L2358)].

aswaterman, 2022-07-21:
It suffices to implement `a u>= b` as `!(a < b)` but with the NaNs masked off:

```
vmfeq.vv    v0, va, va
vmfeq.vv    v1, vb, vb
vmand.mm    v0, v0, v1
vmflt.vv    v0, va, vb, v0.t
vmnot.m     v0, v0
```

And, if `!HONOR_SNANS`, then you can remove the `vmand.mm` by masking the second `vmfeq.vv`:

```
vmfeq.vv    v0, va, va
vmfeq.vv    v0, vb, vb, v0.t
vmflt.vv    v0, va, vb, v0.t
vmnot.m     v0, v0
```

zhongjuzhe, 2022-07-21:
Great！Thank you so much. I will try it today！And I have another question：What about a >= b? This will be generated by -ffast-math，I simpily use vmfle directly，is it sufficient？ ---- Replied Message ----FromAndrew ***@***.***>Date07/22/2022 05:57 ***@***.***> ***@***.******@***.***>SubjectRe: [riscv/riscv-v-spec] How to use RVV instructions to do the codegen for unordered comparison (Issue #814)
It suffices to implement a u>= b as !(a < b) but with the NaNs masked off:
vmfeq.vv    v0, va, va
vmfeq.vv    v1, vb, vb
vmand.mm    v0, v0, v1
vmflt.vv    v0, va, vb, v0.t
vmnot.m     v0, v0

And, if !HONOR_SNANS, then you can remove the vmand.mm by masking the second vmfeq.vv:
vmfeq.vv    v0, va, va
vmfeq.vv    v0, vb, vb, v0.t
vmflt.vv    v0, va, vb, v0.t
vmnot.m     v0, v0

[
{
"target": "https://github.com/riscv/riscv-v-spec/issues/814#issuecomment-1191975134",
"url": "https://github.com/riscv/riscv-v-spec/issues/814#issuecomment-1191975134",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

aswaterman, 2022-07-21:
`>=` should just be `vmfge`.

topperc, 2022-07-22:
For -fno-trapping-math I think there's no requirement that fflags is correct. So `u>=` could be vmflt followed by vmnot. gcc appears to use `u>=` for `!(a < b)` with -fno-trapping-math. vmflt followed by vmnot would be the direct translation of that. So I don't think we're talking about `isless` here.

gcc does have some bizarre behavior with -fno-trapping-math in scalar code. gcc adds a save/restore of fflags for !(a < b) it removes the save restore from `isless` but does not remove it from `!isless` https://godbolt.org/z/17eTaY7G6

closed 2022-07-23 by zhongjuzhe