riscvarchive/riscv-v-spec #815: Question about vl

huikmn, 2022-07-28:
In the riscv-v-spec, vl can be set to a value within 0-vlmax. In order to reduce the hardware complexity, can i add one more contraint to limits vl <= vlen/sew?
So, even VSET instruction sets vlen/sew < vl <vlmax, it still makes vl = vlen/sew. Also rd will be update to vlen/sew。In this case showed in riscv-v-spec works with this extra constraint. 
[image]

I want to ask if this extra constraint is acceptable, will this cause problem of consistency bewteen software and hardware?

aka2en, 2022-07-28:
VLMAX equals to (LMUL * VLEN / SEW). If VL is restricted to (VLEN / SEW), that will lose the benefit of vector register grouping.

huikmn, 2022-07-28:
> VLMAX equals to (LMUL * VLEN / SEW). If VL is restricted to (VLEN / SEW), that will lose the benefit of vector register grouping.

I know it will lose the benefit of register group. But it also reduce the comlexity of hardware. If this policy is acceptable, at least I have one more choice to choose between the benefit of register group and the comlexity of hardware.

aswaterman, 2022-07-28:
No, this would be a nonstandard custom extension, rather than standard V or Zve*.

Making register groups optional would make them unusable by standard software, which would undermine the motivation for the feature.

topperc, 2022-07-28:
Even with that constraint, wouldn't the widening instructions like vwadd still be able to produce a result that required 2 registers to hold the result?

huikmn, 2022-07-28:
> No, this would be a nonstandard custom extension, rather than standard V or Zve*.
> 
> Making register groups optional would make them unusable by standard software, which would undermine the motivation for the feature.

Thank you, i understand.

huikmn, 2022-07-28:
> Even with that constraint, wouldn't the widening instructions like vwadd still be able to produce a result that required 2 registers to hold the result?

yes, it still have the problem, but maybe for this kind of instruction, i can make vlen <= vlen/sew/2, or split the instruction. Either way, it's much simpler than calculating all exception and number of uops for all instructions

closed 2022-07-28 by huikmn