riscvarchive/riscv-v-spec #816: Question about setting vstart

huikmn, 2022-08-01:
According to spec, vstart NORMALLY sets to the position that exception happens. For some out-of-order machine, is it allowed to set vstart earlier than the elements which generates exception.
[image]

For expample, the element0, and element1 are done. element2 and element3 have bank conflict, so element2 and element4 are being processed. then element4 have tlb falut. At this point, element 0,1,2 are finished, element3 is waiting, element4 generates exception. Is it allowed that we set vstart to element3 instead of element4? So i can report the exception earlier.

aka2en, 2022-08-01:
Yes you can, but element3 should still keep the original value.

huikmn, 2022-08-02:
> Yes you can, but element3 has to keep the original value.

ok，thank you