riscvarchive/riscv-v-spec #818: Inconsistent vsetvli policy usage in examples

AdamLatos, 2022-08-11:
In the spec, in `3.4.3`, there are described mandatory flags added to `vsetvli`, with the note:
"The use of vsetvli without these flags is deprecated, however, and specifying a flag setting is now mandatory."

However in the rest of the document `vsetvli` usage is inconsistent, e.g. in `6.0` the policies aren't explicit:
```
 vsetvli rd, rs1, vtypei # rd = new vl, rs1 = AVL, vtypei = new vtype setting
 vsetivli rd, uimm, vtypei # rd = new vl, uimm = AVL, vtypei = new vtype setting
 vsetvl rd, rs1, rs2 # rd = new vl, rs1 = AVL, rs2 = new vtype value
```
`6.1` examples the policies aren't explicit:
```
Examples:
 vsetvli t0, a0, e8 # SEW= 8, LMUL=1
 vsetvli t0, a0, e8, m2 # SEW= 8, LMUL=2
 vsetvli t0, a0, e32, mf2 # SEW=32, LMUL=1/2
```
in `6.4` they are included in the example:
```
vsetvli a3, a0, e16, m4, ta, ma # vtype = 16-bit integer vectors;
```

Do you think it'd make sense to include the flags at least in the monotype examples, since they are mandatory?

aswaterman, 2022-08-11:
There’s a historical element: some of the sample code was written before the deprecation that you mentioned. We’d welcome PRs that fix them!

The examples you mentioned from 6.0 don’t seem immediately relevant: vtypei is pseudocode, not assembly syntax.  Perhaps that should be clarified, but it’s a separate matter.

I agree the examples from 6.1, and any similar ones, should be made conformant.

AdamLatos, 2022-08-11:
Agree about the 6.0 example, didn't realize the policies were a part of `vtype`.

closed 2022-08-12 by AdamLatos