riscvarchive/riscv-v-spec #820: Question about VLMAX

zhongjuzhe, 2022-08-23:
sequence1:
```
li t0, -1
vsetvli x0, t0, e8, m8, ta, ma
....
```

sequence2:
```
vsetvli t0, zero, e8, m8, ta, ma
....
```

Can I assume both sequences are doing the same thing? They are both setting vl to VLMAX? Thanks.

topperc, 2022-08-24:
They should both set the vl to VLMAX. sequence2 is potentially more efficient for hardware. sequence2 allows the hardware to know at instruction decode time that the vl will be VLMAX. sequence1 require the `li` instruction to execute.

nick-knight, 2022-08-24:
They are functionally equivalent w.r.t. `vl`. However, as written, the second sequence writes `t0` with VLMAX, whereas the first sequence writes `-1` to `t0`. In a (contrived) scenario where (a) X-register pressure is very high, (b) VLMAX is not needed to be returned in an X-register, and (c) an unsigned integer greater than VLMAX (like `~0`) is already available in an X-register, the `rs1 != x0` form could be advantageous.

closed 2022-08-24 by zhongjuzhe