riscvarchive/riscv-v-spec #824: Question about vectorize the rounding

zhongjuzhe, 2022-09-18:
Hi, consider the following code:
for (int i = 0; i < n; i++)
  a[i] = round (b[i]);

I wonder do we have the chance to vectorize this code? And How to vectorize this code using RVV ISA ?

Same thing for btrunc, ceil, nearbyint, rint, lrint, lround, lfloor, lceil.
Can anyone teach me about this ? Thanks

jan-wassenberg, 2022-09-19:
Hi, my understanding is that we have to emulate these IEEE-754 operations using other instructions. You can find code to do so here: https://github.com/google/highway/blob/master/hwy/ops/rvv-inl.h#L2957

zhongjuzhe, 2022-09-19:
I still don't get it. Would you mind telling me the exact rvv instruction sequence ?
I want to implement it for auto-vectorization of GCC.

jan-wassenberg, 2022-09-19:
To obtain the RVV instructions, you can search for the function names such as `Abs` in the code, which will lead to a line like
`HWY_RVV_FOREACH_F(HWY_RVV_RETV_ARGV2, Abs, fsgnjx, _ALL)`. The instruction is then v followed by fsgnjx, so vfsgnjx.

Alternatively, there was discussion of this in https://github.com/riscv/riscv-v-spec/issues/623 with some assembly code.

closed 2022-10-08 by zhongjuzhe