riscvarchive/riscv-v-spec #825: vector segment strided store ordering issue

Vivvvvvvv, 2022-09-21:
I was wondering is there a desired accessing order when the segment stride overlaps. As mentioned on the spec that "Accesses to the fields within each segment can occur in any order, including the case where the byte stride is such that segments overlap in memory", which says any order works fine but I just want to double check on this ordering issue in case I missed out anything.  

Considering I have the following setup: for e8, m1, tu, mu, performing the instruction **vssseg2e8.v v8, (x18), x1**, where x18 has the base addr from 00, and x1 has the strided value of 1. Then the expected addr would be:
v8: 00 01 02 03...
v9: 01 02 03 04...
then if the test data are:
v8: a0 a0 a0 a0...
v9: 00 00 00 00.. 
If the desired accessing order is element-vise or horizontal (v8 first then v9), the result would be 0xa0 00 00 00....
However, for vertical order (v8[0] v9[0] v8[1]... respectively), the result would be 0xa0 a0 a0 a0....
I wonder if both result would be considered correct. Hope my question is clear.

aka2en, 2022-09-21:
According to the spec, it looks like what you write. But I'd like the spec to be clearer on the order of "segment", that would help clarify your question.

Vivvvvvvv, 2022-09-22:
> According to the spec, it looks like what you write. But I'd like the spec to be clearer on the order of "segment", that would help clarify your question.

Yea, more clarification is appreciated. Cause two ways will give completely different results which seems too "flexiable".

Vivvvvvvv, 2022-09-22:
Also, FYI, I saw on the spec v0.9, it straightly restrict the access to be element order.(For strided segment stores where the byte stride is such that segments could overlap in memory, the segments must appear to be written in element order.)  But this restriction was removed on spec v1.0.