riscvarchive/riscv-v-spec #826: Does scatter/gather accept negative indices ?

balaji-ch, 2022-09-24:
For instance in `vsuxei16.v vs3, (rs1), vs2, vm # unordered 16-bit indexed store of SEW data`
does vs2 accepts -ve indices like  -1, 0, 1     ?

jnk0le, 2022-09-24:
>If the vector offset elements are narrower than XLEN, they are zero-extended to XLEN before adding to the base effective
address.

balaji-ch, 2022-09-24:
Sorry I didn't get it.

aswaterman, 2022-09-24:
@balaji-ch, the point that @jnk0le was making is that the rule is that addresses are always computed as `rs1 + vs2[idx][IEW-1:0]`, where here IEW is the width of an index element (16 in your example).

So, negative indices are allowed in the sense that they don't cause exceptions, but they probably won't have the effect you intend, since the negative index is turned into a small positive integer.

This concern arises only if the IEW is less than XLEN, so if you were to use e.g. `vsuxei64.v` instead (on an RV64 machine), you'd get the effect you desire.

balaji-ch, 2022-09-24:
@aswaterman
XLEN is 64 for RV64 and only vsuxei64.v/vsoxei64.v work for -ve indexing. Is my understanding correct ?

And I don't see any information like you said about IEW in RVV spec 1.0. Where can I find this ?

aswaterman, 2022-09-24:
I am using the term IEW informally in this context to refer to the EEW of the index element.

You are basically right that you need to use the `ei64` variants on RV64 or the `ei32` variants on RV32 if you want negative indexes to work in the obvious manner.