riscvarchive/riscv-v-spec #827: understanding vector indexed store vsuxei32

balaji-ch, 2022-09-24:
I am trying to understand vector indexed store instructions using spike simulator. Here is the sample code that I tried.

The source array has elements `0xabc0,0xabc1,0xabc2,0xabc3`. With the indexing 3, 2, 1, 0, I expected it to print `0xabc3,0xabc2,0xabc1,0xabc0` but I get 
**Store/AMO access fault!**

But when indexes are given `int32_t indexes[NELMS] = {0,0,0,0};` I get `abc3 0 0 0` as output.  I am unable to understand how the instruction is picking the indexes.

```c
#include <stdio.h>
#include <stdint.h>
#define NELMS 4

void scg(int32_t *dest, int32_t *base_addr, int32_t *offsets, int32_t elements) {
    int32_t vl = 0;
    asm ("vsetvli %0, %1, e32, m1\n" : "=r"(vl) : "r"(elements));
 
    for(int32_t i=0; i<vec_elems;i+=vl){
        asm ("vle32.v v2, (%0)\n" : : "r"(base_addr+i));
        asm ("vle32.v v3, (%0)\n" : : "r"(offsets));
        //asm ("vse32.v v2, (%0)\n" : : "r"(dest+i));
        asm ("vsoxei32.v v2, (%0), v3\n" : :"r"(dest+i));
    }

}

int main() {
	int elements = NELMS;

	int32_t src[NELMS] = {  0xabc0,
                            0xabc1,
                            0xabc2,
                            0xabc3};	                        

	int32_t indexes[NELMS] = {3,2,1,0};
	int32_t dst[NELMS] = {0};

	scg(dst, src, indexes, elements);

	for (int i = 0; i < elements; i++) {
		printf("%x ", dst[i]);
    }

    return 0;

}
```

aswaterman, 2022-09-24:
This is the issue tracker for the ISA spec, and so it's not the appropriate forum to ask for debugging help.

(The Spike issue tracker isn't the appropriate place either, unless you are certain there is a bug in Spike.  I don't think there's evidence of that.  You should consider asking the sw-dev mailing list.)

closed 2022-09-24 by aswaterman

balaji-ch, 2022-09-25:
What is  sw-dev mailing list ? is it google groups ?