riscvarchive/riscv-v-spec #828: Segment fault-only-first load vl setting issue

jinyinghan324, 2022-09-29:
In our design, for segment load instructions, all elements in the first vd are loaded and then the next vd.
For example, for vlseg4e8ff v4, x8. sew = 8 lmul = 1, all elements in v4 are loaded firstly, then all elements in v5 are loaded, then v6 and v7.
If an element > 0 of a middle vd raises an exception, is it correct to reduce vl to the index of element？
In above example, if element 5 of v5 raises an exception, the instruction is stopped. v6 and v7 are not loaded at all. Is it correct to reduce vl to 5？

aswaterman, 2022-09-29:
Correct. The implication is that, if you get a page fault, then after resuming, you’ll re-execute the load starting at element 5, which means you’ll redundantly re-load v5[5].

aswaterman, 2022-09-29:
I misread your description of the implementation.  The execution model is that segments are loaded sequentially: logically, all of segment 0 (i.e. v4[0], v5[0], v6[0], v7[0]) is loaded before segment 1 is loaded.  Following this model, it is correct to set vl=5 if you trap when loading v5[5].

As a separate matter, loading all of v4 before loading any of v5 is inefficient, because it doesn't take advantage of spatial locality in the memory accesses.

closed 2022-12-26 by jinyinghan324