riscvarchive/riscv-v-spec #831: Rounding Floating point to integer - section 13.17

balaji-ch, 2022-10-07:
In 13.17. Single-Width Floating-Point/Integer Type-Convert Instructions
 The conversions use the dynamic rounding mode in **frm**, except for the `rtz `variants, which round towards zero.
 
 How to set the rounding modes ?

balaji-ch, 2022-10-07:
@nick-knight  thank you. I am not interested in Intrinsic but inline assembly.

according to this https://www.andestech.com/wp-content/uploads/Andes-RVV-Webinar-II_final.pdf
there are [2:0] modes in 
[image]
Could you please tell what modes correspond to what values ?
Or is it same like vxrm modes ?

>>I don't understand the rest of your post: what is the perceived issue?
rtz mode is making -ve floats to zero and default rounding is not truncating in spike simulator.

nick-knight, 2022-10-07:
Please refer to Table 13.1 in `riscv-spec.pdf`: https://github.com/riscv/riscv-isa-manual/releases

nick-knight, 2022-10-07:
If negative floats less than `-0.5` are being returned as integer zero, I'd suggest double-checking your assembly code to make sure you are using the signed version of the conversions, rather than the unsigned versions. I just looked at the Spike implementation, and it appears to be invoking SoftFloat for this functionality, so if there is indeed a bug here then please report it.

balaji-ch, 2022-10-07:
> Please refer to Table 13.1 in `riscv-spec.pdf`: https://github.com/riscv/riscv-isa-manual/releases

Thank you. the RVV spec could have linked to this section.

closed 2022-10-07 by aswaterman