riscvarchive/riscv-v-spec #834: What should the result of vfncvt.rod.f.f.w be if the source value is larger than positive/negative maximum value？

jinyinghan324, 2022-10-21:
For example, if sew = 32 and the value is vs2 is larger than positive maximum value of single precision, shoule the result of vfncvt.rod.f.f.w be positive infinite or positive maximum value of single precision?

jinyinghan324, 2022-11-11:
Thank you for your reply

liangkaiwang, 2023-11-16:
@aswaterman sorry to revive this discussion. If this is the intended behavior, i.e. ROD never overflows from normal to infinity during down-conversion, can this be put on to the spec? Just FYI, it is different from other ISA.

aswaterman, 2023-11-16:
Since round-to-odd isn't defined in IEEE 754 like the other rounding modes, I agree this might be appropriate.  Can you prepare a PR?

kdockser, 2023-12-06:
@aswaterman I don't follow why RTO needs to produce max-norm on an overflow in order to convert FP64 to FP16.
 
As I understand it, the only time RTO is required is when the final rounding mode is RNE. In such a case, the max norm in an intermediate precision will result in infinity being returned on the final rounding. It makes no difference if the intermediate RTO returns infinity or max norm.

It seems more intuitive to me that an overflowed value that is subjected to RTO would return infinity. Given a choice on what RTO does on overflow, I would prefer to go with the intuitive approach.

In the case of directed rounding, all stages just need to use the same rounding mode. There is no loss of the sticky bit in directed rounding as it has the same impact whether the guard bit is 1 or not.

Please let me know if I am missing something.

nick-knight, 2023-12-07:
I'm not sure whether there exists a standards document that formally defines odd rounding, as opposed to the five IEEE 754 rounding modes. But every implementation and academic paper I'm aware of follows the definition we've adopted here, i.e., first do `rtz`, then assert the lsb if rounding occurred.

aswaterman, 2023-12-07:
Resolved by e100d4321d7ded558846c32590f9efb0887515da

closed 2023-12-07 by aswaterman

aswaterman, 2023-12-07:
@liangkaiwang The issue is dynamic rounding. Yes, if you statically know the mode is RTZ, then it is correct to use RTZ twice. If you do not know what the rounding mode is, then the situation becomes more complex. (Consider: what code sequence do you need to use if you are statically unaware as to whether the rounding mode happens to be RUP vs. RDN (vs. RNE)?)

By defining ROD in this matter, we already solved this problem elegantly: we can round correctly in two instructions without knowing which dynamic rounding mode was selected.

We do not offer ROD as a standard mode for other operations, so the concern you raised doesn’t apply to the ISA as currently defined. It might be that there is yet another rounding mode, say, ROD_INF, that is more appropriate for general computation. But that’s ultimately a separate matter from the definition of this instruction.