riscvarchive/riscv-v-spec #835: Will vill set to 1 if vl is set to 0?

Vivvvvvvv, 2022-10-25:
Based on spec, it is valid to set vl to 0 using configuration setting instructions. But based on application of vector instructions, setting vl to 0 makes no sense since zero element will be processed. So I'm wondering if vill in vtype register will be set if vl is set to 0? Or what are the expected next-steps if vl is set to 0?

aswaterman, 2022-10-25:
No, setting `vl` to 0 does not cause `vill` to be set to 1.  There is a good reason for this: it eliminates a loop-head branch when vectorizing `for` loops.  The natural vector loop does the right thing even if the trip count is zero.

Vivvvvvvv, 2022-10-26:
Right, I understand. Also, does setting vl to 0 mean that all the dst elements are tail elements? So all of the dst elements should be written based on ta/tu setting.

aswaterman, 2022-10-26:
See section 5.4. Most instructions have no effect at all when vstart >= vl (which includes the case of vstart = 0), and so agnostic elements are not modified.

Vivvvvvvv, 2022-10-26:
I got it. Thanks for the explanation.

closed 2022-10-26 by Vivvvvvvv