riscvarchive/riscv-v-spec #840: RE: Release Version 1.0, Section 13 introduction, FP16 and FP128

FCLC, 2022-12-01:
First, a big thank you to all the designers, maintainers, contributors and anyone else making this possible. 

Context: I'm a younger dev with an HPC + x86_64 AVX512 asm optimization background.

I was reading through the specification now that it's fully published, and a few lines in section 13 caught my attention. Specifically the line 
`
When 16-bit and 128-bit element widths are added, they will be also be treated as IEEE-754/2008-compatible values. 
`
I was slightly disappointed to learn that the base vector ISA would no longer mandate FP16 as a point of conformance with the specification. If this was due to time constraints or a change of direction I'm not certain. 

Thankfully FP16 support seems to be forthcoming via an ISA extension here: https://lists.riscv.org/g/tech-announce/topic/94096689?p=Created%2C%2C%2C20%2C2%2C0%2C0%3A%3Arecentpostdate%2Fsticky%2C%2C%2C20%2C2%2C0%2C94096689%2Cprevid%253D1664819383324599799%2Cnextid%253D1659461430701186831

https://github.com/riscv/riscv-v-spec/issues/5

aswaterman, 2022-12-01:
At the end of the day, whether something is included in the V extension or some other extension matters less than whether a future RISC-V profile mandates those extensions.  It is certainly conceivable that Zvfhmin will be mandated by a RISC-V applications-class profile in the medium term, and Zvfh could additionally be mandated later if demand arises.

I highly doubt that quad-precision will be similarly mandated, because (while it is indeed interesting for some purposes) there just isn't enough demand for it.  After all, lack of demand is why such hardware is hard to come across.

workingjubilee, 2023-02-23:
Once better-known for the "double double" hack, recent implementations of the "Power ISA" (once called PowerPC), starting from POWER9, include "true" IEEE754 128-bit "quad precision" floating-point support as well. So it is not _exactly_ the case that it is "hard to come across", though certainly such machines aren't ubiquitous. Not cheap, either.

aswaterman, 2023-02-23:
Yeah, we're all well acquainted with POWER.  "Hard to come across" was shorthand for "neither ubiquitous nor cheap."

closed 2023-02-23 by aswaterman