riscvarchive/riscv-v-spec #841: Does the V extension depend on the F/D extensions?

jrahmeh, 2022-12-06:
Section 18 of the vector spec sates: 

> The V extension requires the scalar processor implements the F and D extensions ...

Does that imply that the V extension depend on the F/D extensions.  If a CSR instruction clears the V bit in the MISA CSR, is it required to also clear the F and D bits?

Section 3.1.1 of the privileged instruction set states: 

> If an ISA feature x depends on an ISA feature y, then attempting to enable feature x but disable feature y results in both features being disabled. For example, setting “F”=0 and “D”=1 results in both “F” and “D” being cleared.

aka2en, 2022-12-06:
> If a CSR instruction clears the V bit in the MISA CSR, is it required to also clear the F and D bits?

No. You can still have scalar F/D supports after V is cleared in MISA CSR.

aswaterman, 2022-12-06:
Indeed, the dependence is in the other direction.  Clearing misa.F/D must cause misa.V to be cleared, but not vice versa.

closed 2022-12-06 by jrahmeh

deepsrc, 2023-05-31:
If we set misa.F or misa.D to zero, does it result in getting misa.V actually getting set to 0?

OR does it mean that the read of misa.V will return a 0 (and internally it remains set to 1) i.e. if misa.F or misa.D is set back to 1, misa.V will now be set back to 1?