riscvarchive/riscv-v-spec #842: Effecient way to add an integer and floating point ?

sheldon-c73, 2022-12-10:
If we have to add an 8 bit integer to a floating point currently I am extending the 8bit to a 32bit, then converting it into float as follows

```c
        asm ("vle8.v v16, (%0)\n" : : "r"(addr))); // read 8bit int
        asm ("vsetvli %0, %1, e32, m4\n" : "=r"(vl) : "r"(elements)); 
        asm ("vzext.vf4  v4, v16 \n"); // convert to 32 bit
         asm ("vfcvt.f.x.v  v4,  v4\n"); // convert to float
   ```      
         
         is there any alternate way to achieve this to avoid multiple setvlis ?

topperc, 2022-12-10:
You can put the e32 vsetvli before the vle8. vle8 will load 8-bit elements regardless of sew.

topperc, 2022-12-10:
You should also use the x0, x0 form vsetvli when changing vtype if you're not expecting vl to change.

sheldon-c73, 2022-12-10:
> You should also use the x0, x0 form of vsetvli when changing vtype if you're not expecting vl to change.

   you mean like this ?     
        ```c
        asm ("vsetvli %0, %1, e32, m4\n" : "=r"(vl) : "r"(elements));
        ...
        ...
        ...
        asm ("vsetvli %0, %1, e8, m1\n" : "=r"(vl) : "r"(vl));
```

sheldon-c73, 2022-12-10:
> You can put the e32 vsetvli before the vle8. vle8 will load 8-bit elements regardless of sew.

thank you.