riscvarchive/riscv-v-spec #844: Does `Zvfh` imply `Zve32f`?

palmer-dabbelt, 2022-12-16:
This came up in the context of some LLVM ISA string handling discussions <https://reviews.llvm.org/D138807>.  I haven't actually looked super closely, but a few of us got to chatting while reading the ISA spec and I've managed to confuse myself again.

The spec says

    The Zvfh extension requires a standard vector extension with single-precision floating-point support (currently, Zve32f, Zve64f, Zve64d, or V)

which is different than the corresponding `Zfh` language

    The Zfh extension depends on the single-precision floating-point extension, F.

or the `D` language

    The D extension depends on the base single-precision instruction subset F.

That's kind of a subtle difference and I'm not sure how to interpret it.

As some context: We've been assuming `Zfh` implies `F`.  Specifically `riscv64-unknown-linux-gnu-gcc -march=rv64izfh` will generate instructions like `fadd.s`.  As of GCC 11 we also have `D` imply `F` (ie, `riscv64-unknown-linux-gnu-gcc -march=rv64id` emits `fadd.s`).

There's two things I'm having trouble interpreting here:

* Is there a meaningful difference between `requires` and `depends on`, or is it just a prose thing?
* Why are multiple extensions listed?

That second one is kind of tripping me up here, as from reading the spec it's not quite clear why multiple  are listed.  The table (reformatted a bit here for text, so sorry if I'm breaking something) says

    | Extension | Minimum VLEN | Supported EEW |  FP32 | FP64
    | Zve32f    | 32           | 8, 16, 32     |   Y   |  N
    | Zve64f    | 64           | 8, 16, 32, 64 |   Y   |  N

which to me seems to indicate that `Zve32f` is a subset of `Zve64f`.  In that case I'd expect the `Zfh` to just say something like `The Zvfh extension depends on the standard vector extension for single-precision floating-point support Zve32f.` and some additional dependencies between the extensions.

I'm not sure if I'm just reading too much into the wording here, or if there's some difference in behavior that I'm missing?

nick-knight, 2022-12-16:
There are several relations on the set of ISA extensions: containment/inclusion, dependence, requirement, implication, in/compatibility/preclusion. These relations affect the interpretation of the ISA-naming strings as defined in the ISA spec, and the march strings (being) defined in the toolchain-conventions doc. I think these relations need to be clarified more broadly, outside the V-extension spec.

I don't know the answer to the question in the title. My strategy as a developer is to specify both V (or an appropriate Zve*) and Zvfh in the march string. In the case of Zve32f with Zvfh, perhaps the latter implies the former, so my march has some redundancy, but I have bigger fish to fry.

palmer-dabbelt, 2022-12-16:
> I've seen reference to several relations on the set of ISA extensions: containment/inclusion, dependence/requirement, implication, in/compatibility/preclusion. These relations affect the interpretation of the ISA-naming strings as defined in the ISA spec, and the march strings (being) defined in the toolchain-conventions doc. I think these relations need to be clarified outside the scope of the V-extension spec.

Yep, there's a bunch of different terminology used and it frequently gets confusing which results in these sorts of posts ;)

> I don't know the answer to the question in the title. My strategy as a developer is to specify both V (or an appropriate Zve*) and Zvfh in the march string. In the case of Zve32f with Zvfh, perhaps the latter implies the former, so my march has some redundancy, but I have bigger fish to fry.

I definitely agree that the way to deal with this as a user is to just side-step any possible ambiguity by specifying exactly the extensions you want.

aswaterman, 2022-12-16:
The Zfh chapter follows the convention in the unprivileged ISA spec that if extension Foo depends on extension Bar, then Foo implies Bar, and therefore Foo and Foo_Bar are equivalent ISA substrings.  I would support adding explicit text to the unpriv spec that the verb "depend" in this context is an "implies" relation, but that would just be a clarification. 

There are cases where an "implies" relation is not possible.  If Foo requires either Bar or Baz, and neither Bar nor Baz is a superset of the other, then Foo can't imply either--you've got to write either Foo_Bar or Foo_Baz (or Foo_Bar_Baz if Bar and Baz are non-conflicting).  In such cases, we should use the verb "requires".

That concern doesn't apply here, though.  Because Zve32f is a subset of each of Zve64f, Zve64d, and V, it is unambiguous to say "Zvfh implies Zve32f".  We can rewrite that sentence without changing the meaning of the spec.  I'll make a PR.

The Zfh chapter is already saying the right thing (that Zfh and F_Zfh are equivalent), and so no change is necessary there.

PR #845 (2022-12-17): Clarify dependencies between ISA extensions
For consistency with the unprivileged ISA spec, use the verb "depend" to say that an extension requires and implies another.

Explicitly state the dependence between V and Zve*.

Resolves #844
closed 2022-12-17 by aswaterman