riscvarchive/riscv-v-spec #846: Can unmasked stores update tail elements in memory?

palmer-dabbelt, 2022-12-16:
A recent GCC patch <https://inbox.sourceware.org/gcc-patches/20221214113641.63320-1-juzhe.zhong@rivai.ai/> brought up a bit of a grey area in the spec: there's nothing explicitly defining the behavior of tail elements for unmasked vector stores.  I'd always assumed the intent was to have tail elements defined to not manifest as stores (aside from the whole-register flavors), but I've never tried to look this close before.

Not sure if I'm just missing something in the spec here?

nick-knight, 2022-12-16:
It sounds like your concern is that

> Masked vector stores only update active memory elements.

doesn't appear to address the unmasked case. 

I think the fundamental confusion is what it means to be "masked" or "unmasked": I think these terms are used in slightly different ways at different points in the spec. For example, the sentence before the one I quoted says

>  Vector loads and stores are masked [...]

I interpret this as meaning that they have masked forms, where `vm=0` indicates that `v0` is an additional input operand that serves as the mask register.  Another (unrelated) confusion is that some instructions use `vm=0` bit to indicate that `v0` is an additional input operand, but it doesn't play the same role as a mask.

Anyway, my perspective is that the spec intends an equivalence between "unmasked" and "masked with all mask bits asserted". One clue is the choice of `vm=1` (vs. `=0`) to indicate unmasked. Another is the definition of "active" and "inactive":

```
   mask(x)     = unmasked || v0.mask[x] == 1
   active(x)   = body(x) && mask(x)
   inactive(x) = body(x) && !mask(x)
```

aswaterman, 2022-12-16:
Indeed, "masked vector stores only update active memory elements" is a sufficient normative statement.  Tail elements are, by the definition Nick quoted, not active.

aswaterman, 2022-12-16:
Although I think this follows from the definition of "active" and the sentence Nick quoted is not necessary, I'm just going to delete the word "masked" from that sentence to avoid this question going forward.

palmer-dabbelt, 2022-12-16:
My concern was with the "masked" part, not the "active" part, but I think Nick's point about

    Vector loads and stores are masked and do not raise exceptions on inactive elements. 

meaning _all_ vector loads and stores are masked is probably sufficient?  For some reason I'd been reading that as "masked vector loads and stores do not raise exceptions on inactive elements", which is slightly different.

aswaterman, 2022-12-16:
Ack.  I'd still like to clarify this in the spec, even though it's not a normative change, to short-circuit this kind of question in the future.  I'll make a PR shortly.

palmer-dabbelt, 2022-12-16:
WFM.

commit cb46d97c (2022-12-17): Clarify that unmasked stores don't touch tail elts

Resolves #846

As discussed on that issue, this isn't a normative change, but to treat
readers more kindly, we'll remove the emphasis that might have lent the
wrong impression that masked stores have stronger guarantees than
unmasked ones.
closed 2022-12-17 by aswaterman