riscvarchive/riscv-v-spec #849: Question on vector index segment load store

Vivvvvvvv, 2022-12-27:
Hi, i am having a question regarding the constraints on illegal instruction of vector index seg load store. 

Let's say the setting for vtype is e16 m4, and the instruction is vluxseg2ei32.v. For my understanding, under this configuration, the instruction is legal. Because for index load store, I think the "nf*emul <=8 " constraint only applys for the data vector, which in this case emul=lmul=4  and nf is 2.  But if taking index vector into account, the emul would be 8 and the nf*emul will come to 16. I am not sure if I am right on this that this instruction is legal. 

Hope my question is clear.

aswaterman, 2022-12-27:
You are correct; NF only factors into the legality of the data vector.

nick-knight, 2022-12-27:
Another way to think about it is that there is only one index per segment.